motioneye-project/motioneye #611: change motioneye.conf

davidedavlab, 2017-08-02:
How do i modify the motioneye.conf read-only file? I have to change the media_path so that the registrar ends up on a usb keychain I mounted in /home/ftp/storage/usb

Ohem, 2017-08-02:
sudo nano /etc/motioneye/motioneye.conf?

davidedavlab, 2017-08-02:
Welcome to meye-426a93e9!
[root@meye-426a93e9 ~]# sudo /etc/motioneye.conf 
-sh: sudo: command not found

[root@meye-426a93e9 ~]# /etc/motioneye.conf 

                                                                    [ File '/etc/motioneye.conf' is unwritable ]

Ohem, 2017-08-02:
You forgot to include nano, it's the texteditor. sudo gives you root privilegies for that commandline.

davidedavlab, 2017-08-03:
[ File '/etc/motioneye.conf' is unwritable ]

ccrisan, 2017-08-03:
Have you read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ#why-do-i-get-read-only-file-system-when-trying-to-edit-a-file)? Also you might want to edit the [`/data/etc/motioneye.conf`](https://github.com/ccrisan/motioneyeos/wiki/motioneye.conf).

closed 2017-08-03 by ccrisan