motioneye-project/motioneye #612: Doing upgrade - no camera output

rafi1975, 2017-08-04:
Hi,

I've just treied to upgrade motioneye to the latest version on Raspberry Pi.
Upgrade went ok - according to the system. However when the unit is back up online it's still on the old version. And all of a sudden I lost camera output.
Below what I've found in the logs:

[motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-08-04 08:04:12: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-08-04 08:04:13: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-08-04 08:04:15: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-08-04 08:04:16: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-08-04 08:04:17: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2017-08-04 08:04:19: [motioneye]  WARNING: Connect error on fd 21: ECONNREFUSED
2017-08-04 08:04:20: [motioneye]  WARNING: Connect error on fd 21: ECONNREFUSED
2017-08-04 08:04:22: [motioneye]  WARNING: Connect error on fd 21: ECONNREFUSED
2017-08-04 08:04:23: [motioneye]  WARNING: Connect error on fd 21: ECONNREFUSED
2017-08-04 08:04:24: [motioneye]  WARNING: Connect error on fd 21: ECONNREFUSED
2017-08-04 08:04:25: [motioneye]  WARNING: Connect error on fd 21: ECONNREFUSED
2017-08-04 08:04:26: [motioneye]  WARNING: Connect error on fd 21: ECONNREFUSED
2017-08-04 08:04:27: [motioneye]  WARNING: Connect error on fd 21: ECONNREFUSED
2017-08-04 08:04:28: [motioneye]  WARNING: Connect error on fd 21: ECONNREFUSED
2017-08-04 08:04:28: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED
2017-08-04 08:04:29: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED
2017-08-04 08:04:30: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED
2017-08-04 08:04:31: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED
2017-08-04 08:04:33: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED
2017-08-04 08:04:34: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED
2017-08-04 08:04:35: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED
2017-08-04 08:04:36: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED
2017-08-04 08:04:37: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED
2017-08-04 08:04:38: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED

Rafal

ccrisan, 2017-08-04:
I believe you mean motionEyeOS, not motionEye. Did you upgrade from an older version? If yes, you must rewrite the OS image manually.

rafi1975, 2017-08-04:
Hi,

Sorry, I meant motionEyeOS.
Rewrite the image? Does that mean install from scratch?
I'm asking as I have all 7 cameras attached to a house walls and pretty
high. This will require  extra job.

Thanks

ccrisan, 2017-08-04:
What version do you run?

rafi1975, 2017-08-04:
20161212

ccrisan, 2017-08-04:
Damn that's going to be difficult. I also have 10 cameras running motionEyeOS mounted on hardly reachable places, but I use them as fast network cameras and they run a motionEyeOS version from mid 2016. I don't have any reasons to upgrade them. Instead I make sure I run the latest motionEyeOS on my main server that is responsible of all the logic.

rafi1975, 2017-08-04:
I have central hub as well on CentOS but never tried fast network camera
option. Can you still define mask with such option?

ccrisan, 2017-08-04:
You define the mask and pretty much everything on your hub. You add all cameras as network cameras. One thing though is that you must use RPi cameras. If you use USB cameras, the FNC mode is not available.

rafi1975, 2017-08-04:
Can you mix them on the hub e.g. fast network on one RPi and another USB
one on a different RPi?

ccrisan, 2017-08-05:
Of course you can.

closed 2017-08-05 by ccrisan