motioneye-project/motioneye #613: [Feature request] Special frame for action buttons

yanoschik, 2017-08-04:
I still can't get over the limitation of the max count of Action buttons ;) I'd like to have more buttons comfortably available thru the motionEye UI.
+also discussed [here](https://github.com/ccrisan/motioneye/commit/834a68233f8c162b0c77049cbd177ee053876f27)

As it's obvious, that the space inside the camera frame overlay is limited, it would be great to have a separate frame only for 'generic' action buttons - for actions not necessary bound to a specific camera.
My idea is that this could be a simple frame (just like another camera frame), optionally enabled via checkbox in Preferences. I would host buttons (no special graphics/icons needed) generated by a similar mechanism - loaded executable scripts placed in it's own folder (say <[configuration folder](https://github.com/ccrisan/motioneye/wiki/Configuration-File#conf_path)>/actions/). The script name would become the button title, while a prefix defines the order (e.g. script "01_Run_My_Command" --> first button named "Run My Command").
Thanks

Matanuska, 2018-08-15:
I also need this feature, in the header bar for example
[image]

n3roGit, 2020-07-23:
yes please

MichaIng, 2024-05-28:
Fixed with: #2903

PR #2903 (2024-05-28): Increase amount of action buttons 
Hi

I made some UI changes to increase the available action buttons. There have been several requests concerning this. This pull request fixes: #613 ,#844, #1766. 

[image]
[image]
[image]
[image]

I moved the frame rate and monitoring info to the top. I tried to optimize everything for space constraints. There are two rows in the top left. First one for camera name and frame rate and the second one for monitoring info. If there is no monitoring info, the second row disappears and name and fps are centered. The bottom is now solely for action buttons. Instead [... cut]
closed 2024-05-28 by MichaIng