motioneye-project/motioneye #617: RPi camera auto brightness not working

nabh, 2017-08-06:
HI, I'm running version 0.36.1 with motion 4.0.1 with Jessie latest version and all apt-get's are upto date. All starting from clean installs with known good hardware.

I have 3 setups like this: 
1 x RPi 2 with RPi camera 
1 x RPi 2 with usb camera
1 x RPi Zero W with RPi camera

Problem: With the RPi's with Rpi Cameras ONLY the usb set up is fine. After the cameras have been looking at a low light image over night or dark room the image stays dark even after full day light or room light is switched  on.  To all appearances the "auto" function of the camera is not working.

After investigation I find that:
1. Effectively the cameras no longer auto adjust the brightness
2. I have auto brightness turned OFF in he web UI
3. I can restore the image to a better exposure by toggling one of the settings which does not need to be related to exposure, and clicking the apply button. E.g. Changing the camera name also worked.
4. AUTO BRIGHTNESS CONTROL: Starting with the setting in the auto brightness OFF position, toggle the control to show ON (DO NOT click apply), then toggle the control to show OFF again, now click apply. The result is that the camera shows a well exposed image at least until then light level changes. Auto exposure appears to be in one shot mode.

I hope this helps either reproduce the problem or point to a solution.

Neale

ccrisan, 2017-08-07:
The autobrightness option is know not to work with cameras that have some kind of autobrightness mechanism themselves. There is nothing motionEye can do about this.

closed 2017-08-07 by ccrisan

nabh, 2017-08-07:
Please reopen and reread the report. 

I know auto brighness does not work in motioneye with auto brightness capable cameras like the RPi camera. I stated above that "I have auto brightness turned OFF in he web UI"

I only used it as a way to kick the UI I never actually turn it on.

Something in motioneye is preventing the camera from using it's own auto mechanism.

I've also checked that the camera is still working correctly withoit motioneye running by taking a movie and transitioning from dark to light and back. The exposure works fine.

ccrisan, 2017-08-07:
I read the issue and I understood your problem. Again, there's nothing motionEye can do about it. Keep in mind that motionEye is just a frontend for motion and that it has very little to do with the actual video capture and processing.

You may want to open a motion issue instead, they might have a solution for you.

nabh, 2017-08-07:
Understood but I've been using motioneye for 2 years now and this is the first time I have seen this issue usually it works fine.I only recently upgraded the set up to the latest version(0.36.1) and that's when the problems started. So either one of the dependencies is causing the problem or motioneye is inadvertently turning OFF the camera auto exposure function. How do I roll back motioneye to a pervious version?

ccrisan, 2017-08-07:
I doubt the new version of motionEye is causing these problems. Have you also updated motion? Or maybe the entire OS with kernel, drivers, camera firmware and whatnot? 

If you're certain that things worked well before the upgrade, you might need to try to restore the old environment and upgrade components one by one and see which one is causing the problem.

wborck, 2017-10-27:
Hello, is it possible to set "Automatic Brightness" to "ON" / "OFF" with curl and then save it with "Apply"?

Greetings,
wborck