motioneye-project/motioneye #618: tornado.httpclient not found after upgrade to newest version

Samfox2, 2017-08-08:
Hi,

I upgraded motioneye to the newest version but now I`m getting the following error message:
from tornado.httpclient import AsyncHTTPClient
ImportError: No module named tornado.httpclient

I already tried to re-install tornado but it always tells me that I allready have the most recent version installed. 

Does anyone have the same issue?

ccrisan, 2017-08-08:
Do you run it using python2 or python3?

Samfox2, 2017-08-08:
I`m using python2

ccrisan, 2017-08-08:
If you run `python2` and then type `import tornado.httpclient`, does it give an error?

Samfox2, 2017-08-08:
Does'nt find the module. But pip tells me that it is allready installed...

Samfox2, 2017-08-08:
Solved it. The problem was a sym link to an old python version.

closed 2017-08-08 by Samfox2