motioneye-project/motioneye #619: [Feature Request] Prebuilt image for embedded x86 systems (like motionPie)

lugiank, 2017-08-08:
Is it possible to provide a stripped down linux distro with everything set up like MotionPie but for x86 systems?
Me not being a linux user makes it hard to manage and troubleshoot my surveillance system.

ccrisan, 2017-08-08:
There are no plans for a prebuilt motionEyeOS image for x86. The reason is the diversity of the available hardware and configurations and the fact that most x86 machines are powerful machines on which users normally have other stuff installed.

If you have a particular x86 board in mind, that's another story and we could evaluate the possibility of porting motionEyeOS to that board.

lugiank, 2017-08-08:
I was thinking about running as a service in windows a separate VM hosting motioneye.

Since it is pretty hard working with the ubuntu server version because of no GUI, i installed ubuntu in the VM and added motioneye, but having a stripped down linux distro with motioneye only would be much less resource intensive

All of this because my raspberry can't handle more than 2 hd cameras on it's own

Thank you for your patience

(An auto-install script would also do the trick)

ccrisan, 2017-08-09:
Well that should be solved with a Dockerfile or an auto-install script, as you mentioned. The overhead of maintaining yet another prebuilt OS image is too big.

closed 2017-08-09 by ccrisan