motioneye-project/motioneye #620: How to get a JSON command to run with 'Run a command' ?

D3rax, 2017-08-08:
Hi,

I'm enjoying Motioneye very much. I got 2 camera's setup and I have notifications through email working when motion is detected.

I would also like to run a command when motion is detected. I want this command to switch on a switch in my Domoticz setup.

At the moment I have this command entered in the 'Run a command' section:

http://<domoticz IP>:8080/json.htm?type=command&param=switchlight&idx=447&sw

When I enter this command in my browser, the switch is turned on in Domoticz. But I looks like Motioneye does not run the command or is not able to run it?

I have no idea why this is not working? I hope somebody can help me out!

Cheers,

Xander

D3rax, 2017-08-08:
The above json command is not complete, this is the command i'm using:

http://000.000.0.0:8080/json.htm?type=command&param=switchlight&idx=447&sw

ccrisan, 2017-08-08:
That's not a command and that is definitely not a valid IP address. You're looking for Motion Notifications > Call A Webhook.

closed 2017-08-08 by ccrisan

D3rax, 2017-08-09:
Thanks for you reply! 

The above IP adres is not the IP adres I used, I replaced the original adres with 0's. 
I will try the Call a Webhook option.