motioneye-project/motioneye #623: No camera image

gyurobenjamin, 2017-08-14:
On Raspberry zero w with onboard camera, tried to use motionEye.
Finally I could add my onboard camera but still no image.

Is there a way to get more information about this problem? Maybe anyone has any idea?

ccrisan, 2017-08-14:
Have you read https://github.com/ccrisan/motioneye/wiki/Troubleshooting#raspberry-pi-camera-module-not-detected?

gyurobenjamin, 2017-08-14:
No, I haven't before. Most of the list is OK in my case. I'm not sure in the GPU, later I will check it. I can capture a photo with raspistill, and my camera is not genuine. How can I make sure my camera is compatible with motioneye or any other software?

ccrisan, 2017-08-14:
Well if the camera works with Raspbian and the raspistill command, it should work just fine with motionEye.

gyurobenjamin, 2017-08-26:
I tried again with the normal image instead of the minimal version. With this everything working fine.

closed 2017-08-26 by gyurobenjamin