motioneye-project/motioneye #624: Nginx: Add websites that works at the same time with Motioneye

chrissierra, 2017-08-14:
Remember this:

Why Nginx?

Nginx is configurable, fast and easy to install. Running behind the Nginx webserver (using Nginx as a reverse proxy) can be useful when:

you want to use HTTPS with motionEye
you want to serve multiple websites (including motionEye) on the same webserver (accessible at the same IP address)
you want to take advantage of various Nginx modules (such as rate limiting or HTTP authentication)

//*****************************************************************************

I want to serve multiples websites, and explication above is insufficient. Can somebody give some better explanation?

chrisf4lc0n, 2017-08-15:
Ngnix is just one of many you can use if it is not suitable for you go with Apache. I personally use Lighttpd and it does everything I want it to do including NextCloud...

ccrisan, 2017-08-15:
This is not a valid motionEye issue. If you want to learn how to use nginx, there are plenty of articles online. If you are looking for help from the community, you can head over to https://groups.google.com/forum/#!forum/motioneye.

closed 2017-08-15 by ccrisan