motioneye-project/motioneye #626: MotionEye on Armbian

chrissierra, 2017-08-16:
Can I install MotionEye on Armbian? (Orange Pi)

Thanks!

chrisf4lc0n, 2017-08-16:
Why not? As long as you meet all the software requirements I cannot see the reason why wouldn't it work... I personally ran it on Banana Pi M1 and it was fine, but it could not detect the internal camera.

chrissierra, 2017-08-16:
But about the installation setup that are on https://github.com/ccrisan/motioneye/wiki/Installation ; what must I to use? Install On Other Distros ? or Install On Ubuntu?

chrisf4lc0n, 2017-08-17:
If it is Ubuntu you have installed just follow the instructions for Ubuntu.

tehKaiN, 2018-01-01:
I've used MotionEye on Armbian with success on OrangePi One. It's best to install MotionEye using `pip` - this way you get newest and updatable version. Before that install Motion separately - I recommend going through manual compiling - build procedure is clearly documented on project's page.

ccrisan, 2018-01-01:
@tehKaiN is right - motionEye normally works on a variety of distros using `pip` and the install instructions should cover common setups.

On a side note, the next prerelease of [motionEyeOS](https://github.com/ccrisan/motioneyeos) will add OrangePI One support.

closed 2018-01-01 by ccrisan

UrbanSide, 2020-08-15:
Hello,how i can install MotionEye on OPI Lite? My os it last Armbian OS