motioneye-project/motioneye #627: Mask and Threshold

jasonhpchu, 2017-08-18:
I'm trying to figure out what value I should set my Threshold at.
So I have a 1920x1080 stream.
Without the mask, threshold is just percentages to the total pixels, which would be 2,073,600 in this case.
Now with mask, I'm guessing the total pixels changed number is subtracted by any pixels changes within the masked off square area?  Then the percentage threshold is still to the unmasked resolution total pixels?

I'm guessing my question is, how do I find out the maximum number of pixels that could be changed after applying the mask.  That way I can tweak the Threshold to be a percentage the max (if I set it above that max, then it'll never trigger).

Thanks.

richri, 2018-02-18:
AFAIK the threshold is not affected by the mask. The number of pixels stays the same, only due to the mask, certain parts of the image are not evaluated for motion detection.

Please correct me, if I am wrong.

Best regards

kurazli, 2019-04-12:
I am trying to find out the same.
Anyone knows any details or has a hint?

zhangziran, 2023-11-18:
I am trying to find out the same.
Anyone knows any details or has a hint?

zagrim, 2023-11-19:
The percentage used for setting threshold in MotionEye is always the percentage of the total frame size. It is just a simple representation of the actual setting passed to Motion which is given in number of pixels.

[Motion documentation](https://motion-project.github.io/motion_config.html#threshold) says about threshold:
> The threshold is the number of changed pixels counted after noise filtering, masking, despeckle, and labelling.

So, masked areas do not result in any changed pixels, but since they are not taken into account in the threshold percentage setting, one should just mentally take masked areas into account when estimating a good threshold value (i.e. what is are size of the frame area where motion can occur that you want to capture). In my experience it's anyway more about trial and error to see what a proper threshold should be, unless the setup is somehow such that the actual areal size of the expected movement is possible to define with even some accuracy.

JamBalaya56562, 2026-11-12:
zagrim's answer is right: motionEye's percentage is always of the whole frame, `width × height`, and is passed to Motion as a number of pixels. Motion counts changed pixels after masking, so masked pixels never count; to keep the same sensitivity with a mask, lower the percentage by the masked share of the frame. Closing as answered.

closed 2026-11-12 by JamBalaya56562