motioneye-project/motioneye #628: Digist Authentification

menschmachine-de, 2017-08-18:
Hi,

I would like to use motioneye with a Dahua IP Camera

VLC plays nice.
The Dahua has only Digist-Auth .
Does motioneye support Digist-Auth.

Could find anything
Best regards

ccrisan, 2017-08-19:
Yes, it does, if you're talking about adding an IP camera that requires digest authentication. Make sure you're using motion-4.x.

dec0dOS, 2019-09-13:
Getting same error with Dahua IP camera. 
[image]

WarriorXK, 2019-09-14:
Works for me when using RTSP like so: rtsp://192.168.1.51:554/live

dec0dOS, 2019-09-14:
RTSP is too CPU consuming so JPEG is more preferable.

JamBalaya56562, 2026-11-10:
Closing, as this was answered: Motion 4.x handles digest authentication for network cameras, and the Dahua camera worked here with its RTSP URL (`rtsp://<ip>:554/live`). Please open a new issue with `motion.log` if a camera still fails to authenticate with a current motionEye and Motion.

closed 2026-11-10 by JamBalaya56562