motioneye-project/motioneye #629: Capture long Videos

c2ho, 2017-08-19:
Hey there.

I need to capture long vids with my RPi3 + official cam v2.1, because i got an idiotic neighbor.
Found this project yesterday and tried to get it working for my needs.

The main problem is, that recording is cancelled after some time and the stream doesn't show a picture, too. So, recording videos to my NAS is getting aborted for any reason. Sometimes the stream is up again and new files can be found at the NAS.

What i need:
Full video recording from 4 AM - 10 PM (in one file if that's possible)
Audio recording (is there a chance to do that?)

The RPi3 is connected via WiFi and LAN, i think the wired connections gets preferred by the system itself. I can access the WebUI via WiFi and LAN (no signal on both).

Logs are attached to this post.

[motionEyeOS_logs_WebUI_c2ho.zip](https://github.com/ccrisan/motioneye/files/1236522/motionEyeOS_logs_WebUI_c2ho.zip)

chrisf4lc0n, 2017-08-20:
Can you change the log level of MotionEye to debug, please?

c2ho, 2017-08-20:
Now with "Debugging" enabled in WebUI 
[motionEyeOS_logs_Debugging_WebUI_c2ho.zip](https://github.com/ccrisan/motioneye/files/1236894/motionEyeOS_logs_Debugging_WebUI_c2ho.zip)

chrisf4lc0n, 2017-08-20:
2017-08-20 06:18:33: [motioneye]    ERROR: failed to mount smb share "//192.168.0.102/Datengrab" at "/data/media/motioneye_192_168_0_102_datengrab_christian"
The samba mount seems to be dropping.
Could be the unstable WiFi connection...

c2ho, 2017-08-20:
So it would be better to connect just via LAN? How do I set up the network space with that 

Another problem is, that the RPi3 is shutting down - or I just can't reach it via browser. 
Pinging with another device gets a positive response on both IPs, LAN and WiFi.

chrisf4lc0n, 2017-08-20:
Are you running MotionEye OS or MotionEye?

c2ho, 2017-08-20:
Ehm, MotionEyeOS I think.
As I see this Repo is for MotionEye, not the OS... :[

chrisf4lc0n, 2017-08-20:
Not to worry... WebUI might be unusable due to many reasons, in my case it kept dropping when I was trying to use the same stream port as for my Calibre service. Can you SSH to the Pi?

c2ho, 2017-08-20:
Yes, I can access it via SSH using root.

chrisf4lc0n, 2017-08-20:
Check what is the space left on the SD card and if there are any files in /var/lib/motioneye

c2ho, 2017-08-20:
The SD Card got minimum 14GB space, 
At my system, there is no "motioneye" in /var/lib/ just "motioneye.pid" and "motion.pid"
[image]

chrisf4lc0n, 2017-08-20:
OK, so you are not running out of space on the SD card, nor MotionEye saves files locally. Since you have got 10GB space can you make it to save files locally and see if the problem persists? I do however put my bet on erratic WiFi.

c2ho, 2017-08-20:
I would try that, but I can't access the WebUI to say that it should save ~1h on thr SD card. Now it is set to 24h to NAS.

Even a restart of the RPi3 won't bring up the WebUI, but I still can reach it through SSH and ping the RPi3.

chrisf4lc0n, 2017-08-20:
That looks like a pickle to me, a re-installation might be needed to further investigate the problem if that still appears after fresh install.

c2ho, 2017-08-21:
I reinstalled MotionEyeOS 15 minutes ago, keeping WiFi deactivated, just using LAN with 1 GBit/s recording straight to the NAS. I am waiting for the next 'crash'. After that I will change the path from NAS to local savings. If that won't work... Yeah, I don't know what I will do then.

If it keeps 'crashing', I will upload the debugging logs again.

c2ho, 2017-08-21:
It seems like there is a problem with SMB / network. Recording is running for about 1h now without any problems using the SD card als storage device.
I don't know what the problem is because the RPi3 is connected via LAN, not WiFi.

Is there a way to figure out what the problem is with saving the files right to my NAS? The files are getting huge so I don't want to copy them manually to my NAS every day - if it's realizable with FTP etc. that would be OK, too. I can set up a FTP server at my NAS, but saving straight to a shared folder would be better.

chrisf4lc0n, 2017-08-23:
At least you got to the bottom of it. I doubt it is the problem on the MotionEyeOS side, your NAS seems to be playing up. Try to test it from another device with a looped up some kind of NAS speed tester or command line from Linux.

c2ho, 2017-08-24:
I tested the NAS drive (over LAN) with CrystalDiskMark, Screenshot below. 
My NAS OS shows me, that I have SMART errors at this drive (it's a single one, no RAID at the moment). I will change the path to another drive and test it again. 

Do you have any ideas how I can record sound next to a video with MotionEyeOS? 
I've found [this](http://www.lavrsen.dk/foswiki/bin/view/Motion/SoundAudioRecording) but I can't set up a Raspbian with MotionEye at the moment (installation brings up some problems). AND I need to record the sound completely, not triggered by any motion detection.
Bought an USB Soundcard that works under Linux without installing special drivers and got a cheap mic for it with good sound quality. 
If you have any other suggestions to set up the stuff I want to do, just tell me and I will try that.

[image]

gregewing, 2017-09-18:
I suggest putting raspian on a different SDCard and installing iperf.  se iperf to test the network performance to and from your RPi in its intended location.

MichaIng, 2026-10-12:
I'll close this old issue. Let us know if there are issues when running latest motionEye (not motionEyeOS) on the Raspberry Pi distro of your choice.

closed 2026-10-12 by MichaIng