motioneye-project/motioneye #630: Dead camera makes MotionEye unusable

chrisf4lc0n, 2017-08-19:
As in the subject.
Just recently one of my cameras died and rendered the MotionEye WebGUI completely unusable until I deleted it.

The Debug Log file for that looks lie that in a loop:
Aug 19 19:54:49 Ubunciak meyectl[2153]:  WARNING: Connect error on fd 21: ECONNREFUSED
Aug 19 19:54:49 Ubunciak meyectl[2153]:    DEBUG: connection closed for mjpg client for camera 2 on port 8082
Aug 19 19:54:49 Ubunciak meyectl[2153]:    DEBUG: mjpg client for camera 2 on port 8082 removed
Aug 19 19:54:49 Ubunciak meyectl[2153]:    DEBUG: creating mjpg client for camera 5
Aug 19 19:54:49 Ubunciak meyectl[2153]:  WARNING: Connect error on fd 21: ECONNREFUSED
Aug 19 19:54:49 Ubunciak meyectl[2153]:    DEBUG: connection closed for mjpg client for camera 5 on port 8085
Aug 19 19:54:49 Ubunciak meyectl[2153]:    DEBUG: mjpg client for camera 5 on port 8085 removed
Aug 19 19:54:49 Ubunciak meyectl[2153]:    DEBUG: creating mjpg client for camera 2

In the same time the WebUI constantly starts and stops all the camera feeds making it completely unusable.
An easy fix to that would be to delete the camera after certain number of retries from MotionEye side or certain time.

ccrisan, 2017-08-19:
When you see that ECONNREFUSED, there's something wrong going on with motion. It would be interesting to see what the motion log says about this.

chrisf4lc0n, 2017-08-19:
motion.log from the moment the faulty camera gets connected:

[2:ml2] [NTC] [NET] netcam_rtsp_resize_ntc:
[2:ml2] [NTC] [NET] netcam_rtsp_resize_ntc: ****************************************************************
[2:ml2] [NTC] [NET] netcam_rtsp_resize_ntc: The network camera is sending pictures in a different
[2:ml2] [NTC] [NET] netcam_rtsp_resize_ntc: size than specified in the configuration file.
[2:ml2] [NTC] [NET] netcam_rtsp_resize_ntc: The picture is being transcoded into the size
[2:ml2] [NTC] [NET] netcam_rtsp_resize_ntc: requested in the configuration.  If possible change
[2:ml2] [NTC] [NET] netcam_rtsp_resize_ntc: netcam or configuration to indicate the same size
[2:ml2] [NTC] [NET] netcam_rtsp_resize_ntc: to possibly lower CPU usage.
[2:ml2] [NTC] [NET] netcam_rtsp_resize_ntc: Netcam: 1280 x 720 => Config: 1920 x 1080
[2:ml2] [NTC] [NET] netcam_rtsp_resize_ntc: ****************************************************************
[2:ml2] [NTC] [NET] netcam_rtsp_resize_ntc:
[2:ml2] [DBG] [NET] netcam_image_read_complete: Calculated frame time 60685.597656
[2:ml2] [NTC] [NET] netcam_connect_rtsp: Camera connected
[2:ml2] [DBG] [NET] netcam_image_read_complete: Calculated frame time 55099.039062
[2:ml2] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[2:nc2] [ALR] [NET] netcam_handler_loop: Camera handler thread [8] started
[2:ml2] [NTC] [STR] http_bindsock: listening on 127.0.0.1 port 8084
[2:ml2] [NTC] [ALL] motion_init: Started motion-stream server on port 8084 (auth Disabled)
[2:ml2] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 40 items
[2:nc2] [DBG] [NET] netcam_image_read_complete: Calculated frame time 50442.136719
[3:ml3] [INF] [NET] netcam_proc_jpeg: processing jpeg image - content length 39420
[2:nc2] [DBG] [NET] netcam_image_read_complete: Calculated frame time 45874.722656
[5:ml5] [DBG] [NET] netcam_check_buffsize: expanding buffer from [0/0] to [0/462848] bytes.
[5:ml5] [DBG] [NET] netcam_check_buffsize: expanding buffer from [0/0] to [0/462848] bytes.
[2:nc2] [DBG] [NET] netcam_image_read_complete: Calculated frame time 41806.949219
[0:av5] [ERR] [ENC] ffmpeg_avcodec_log: mb_type 47 in P slice too large at 13 5
[0:av5] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 13 5
[0:av5] [INF] [ENC] ffmpeg_avcodec_log: concealing 3236 DC, 3236 AC, 3236 MV errors in P frame
[2:nc2] [DBG] [NET] netcam_image_read_complete: Calculated frame time 38129.855469
[2:nc2] [DBG] [NET] netcam_image_read_complete: Calculated frame time 34798.171875
[0:av5] [INF] [ENC] ffmpeg_avcodec_log: Reinit context to 640x368, pix_fmt: yuv420p
[0:av5] [ERR] [ENC] ffmpeg_avcodec_log: P sub_mb_type 5 out of range at 12 10
[0:av5] [ERR] [ENC] ffmpeg_avcodec_log: P sub_mb_type 7 out of range at 2 5
[0:av5] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 12 10
[0:av5] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 2 5
[0:av5] [DBG] [ENC] ffmpeg_avcodec_log: Invalid UE golomb code
[2:nc2] [DBG] [NET] netcam_image_read_complete: Calculated frame time 31799.955078
[0:av5] [INF] [ENC] ffmpeg_avcodec_log: concealing 2837 DC, 2837 AC, 2837 MV errors in P frame
[0:av5] [INF] [ENC] ffmpeg_avcodec_log: concealing 3247 DC, 3247 AC, 3247 MV errors in P frame
[2:nc2] [DBG] [NET] netcam_image_read_complete: Calculated frame time 29188.958984
[0:av5] [INF] [ENC] ffmpeg_avcodec_log: concealing 2626 DC, 2626 AC, 2626 MV errors in P frame
[2:nc2] [DBG] [NET] netcam_image_read_complete: Calculated frame time 26875.164062
[2:nc2] [DBG] [NET] netcam_image_read_complete: Calculated frame time 24608.546875
[2:nc2] [DBG] [NET] netcam_image_read_complete: Calculated frame time 22658.091797
[2:nc2] [DBG] [NET] netcam_image_read_complete: Calculated frame time 20765.683594
[2:nc2] [DBG] [NET] netcam_image_read_complete: Calculated frame time 19063.314453
[0:av5] [ERR] [ENC] ffmpeg_avcodec_log: out of range intra chroma pred mode at 11 5
[0:av5] [ERR] [ENC] ffmpeg_avcodec_log: out of range intra chroma pred mode at 49 9
[0:av5] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 11 5
[0:av5] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 49 9
[0:av5] [INF] [ENC] ffmpeg_avcodec_log: concealing 3238 DC, 3238 AC, 3238 MV errors in P frame
[0:av5] [INF] [ENC] ffmpeg_avcodec_log: concealing 2880 DC, 2880 AC, 2880 MV errors in P frame
[2:nc2] [DBG] [NET] netcam_image_read_complete: Calculated frame time 17533.283203
[2:nc2] [DBG] [NET] netcam_image_read_complete: Calculated frame time 16151.854492
[2:nc2] [DBG] [NET] netcam_image_read_complete: Calculated frame time 14907.569336
[3:nc3] [INF] [NET] netcam_check_content_type: Content-type image/jpeg
[3:nc3] [INF] [NET] netcam_check_content_length: Content-Length -1
[3:nc3] [INF] [NET] netcam_check_content_length: Content-Length 39596
[3:nc3] [INF] [NET] netcam_read_next_header: Found image header record
[2:nc2] [DBG] [NET] netcam_image_read_complete: Calculated frame time 13890.612305
[3:nc3] [DBG] [NET] netcam_image_read_complete: Calculated frame time 94915.046875
[3:ml3] [INF] [NET] netcam_init_jpeg: ***new pic delay successful***
[3:ml3] [INF] [NET] netcam_init_jpeg: jpeg_error 0
[2:nc2] [DBG] [NET] netcam_image_read_complete: Calculated frame time 12957.151367
[3:ml3] [INF] [NET] netcam_image_conv: jpeg_error 0
[0:av5] [ERR] [ENC] ffmpeg_avcodec_log: P sub_mb_type 5 out of range at 12 5
[0:av5] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 12 5
[... cut ...]
[5:ml5] [DBG] [NET] netcam_image_read_complete: Calculated frame time 405255.125000
[5:ml5] [NTC] [ALL] image_ring_resize: Resizing pre_capture buffer to 1 items
[5:nc5] [ALR] [NET] netcam_handler_loop: Camera handler thread [9] started
[5:nc5] [DBG] [NET] netcam_image_read_complete: Calculated frame time 364830.406250
[5:ml5] [NTC] [STR] http_bindsock: error binding on any IPv4 address port 8081
[5:ml5] [ERR] [ALL] motion_init: Problem enabling motion-stream server in port 8081: Address already in use
[5:ml5] [NTC] [ALL] motion_loop: Thread exiting

ccrisan, 2017-08-19:
Somehow your thread-*.conf files got messed up in a way that the same streaming port is used by multiple cameras. Please take a look over all your thread-*.conf files and make sure the `stream_port` is distinct in each of them. It should normally be 8080 + the camera id.

chrisf4lc0n, 2017-08-20:
I got to the bottom of it, thanks for pointing me in the right direction...
It looked like I have had Calibre running on the 8081, the same one as it would be for the new camera added. By changing the Calibre port 8079 I got everything back. It also turns out that the camera is not faulty at all.
One thing makes me wonder however, I have ran Calibre, parallel to this camera for over a year and it has never been an issue, even though they used the same ports and I have never enabled streaming...

closed 2017-08-20 by chrisf4lc0n