motioneye-project/motioneye #632: No video after ugprade to raspbian stretch

rscoleNZ, 2017-08-19:
Firstly, I am very new to linux/github etc, so I will do the best I can.
I have upgraded from Jessie to Stretch on my pi3.
I primarily use a usb webcam with motioneye, and everything worked fine on Jessie.

As best I can tell this is from the error log? when I ran journalctl -u motioneye and motion I got a lot of the below.

<code>Aug 20 09:46:14 raspberrypi meyectl[2722]:    ERROR: motion not running, starting it
Aug 20 09:46:14 raspberrypi meyectl[2722]:    ERROR: failed to start motion: motion failed to start
Aug 20 09:46:14 raspberrypi meyectl[2722]: Traceback (most recent call last):
Aug 20 09:46:14 raspberrypi meyectl[2722]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/server.py", line 323, in checker
Aug 20 09:46:14 raspberrypi meyectl[2722]:     motionctl.start()
Aug 20 09:46:14 raspberrypi meyectl[2722]:   File "/usr/local/lib/python2.7/dist-packages/motioneye/motionctl.py", line 134, in start
Aug 20 09:46:14 raspberrypi meyectl[2722]:     raise Exception('motion failed to start')
Aug 20 09:46:14 raspberrypi meyectl[2722]: Exception: motion failed to start
</code>

<code>Aug 20 08:18:48 raspberrypi systemd[1]: Starting LSB: Start Motion detection...
Aug 20 08:18:49 raspberrypi motion[330]: Not starting motion daemon, disabled via /etc/default/motion ... (warning).
Aug 20 08:18:49 raspberrypi systemd[1]: Started LSB: Start Motion detection.
Aug 20 09:24:09 raspberrypi systemd[1]: Stopping LSB: Start Motion detection...
Aug 20 09:24:09 raspberrypi motion[12519]: Stopping motion detection daemon: motionNo /usr/bin/motion found running; none killed.
Aug 20 09:24:09 raspberrypi motion[12519]: .
Aug 20 09:24:09 raspberrypi systemd[1]: Stopped LSB: Start Motion detection.
Aug 20 09:24:12 raspberrypi systemd[1]: Starting LSB: Start Motion detection...
Aug 20 09:24:13 raspberrypi motion[12599]: Not starting motion daemon, disabled via /etc/default/motion ... (warning).
Aug 20 09:24:13 raspberrypi systemd[1]: Started LSB: Start Motion detection.
</code>

I have used cheese and that confirms the webcam does work on Stretch.
Any help is appreciated. 
Also, as it looks like the fault is with motion, do I need to copy and past this to the motion github?

ccrisan, 2017-08-19:
You should probably open a motion issue. Copy-pasting stuff from motionEye doesn't make any sense to them, but you could try to manually start motion and see what's going on. Log files created by motion will help them investigate the issue.

closed 2017-08-19 by ccrisan

rscoleNZ, 2017-08-20:
As I could not manually start motion, I managed to resolve this myself by doing the following (though I am certain I had already tried this). 

apt-get remove motion
apt-get autoremove
apt-get install motion

Thanks