motioneye-project/motioneye #635: Raspbian (based on Debian Stretch) unresolved dependencies

mradeschnig, 2017-08-21:
On the new release of Raspbian (based on Debian Stretch) some packages are not available from step 2. of you installation procedure recommended for Raspbian.

DawsonLeBrown, 2017-08-21:
I too am having trouble with the latest Raspbian build.

Following the published instructions, then when installing dependencies from the repositories, it errors. The first dependency it errors on is libx264-142. I tried installing the dependencies one at a time until I found the first that errors:

> pi@raspberrypi:~ $ sudo apt-get install libx264-142
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> E: Unable to locate package libx264-142
> 

There appear to be two libx264 packages available: libx264-148  libx264-dev

libavcodec56 gives a different error:

> pi@raspberrypi:~ $ sudo apt-get install libavcodec56
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> Package libavcodec56 is not available, but is referred to by another package.
> This may mean that the package is missing, has been obsoleted, or
> is only available from another source
>
>E: Package 'libavcodec56' has no installation candidate

libavformat56 similarly:

> pi@raspberrypi:~ $ sudo apt-get install libavformat56
> Reading package lists... Done
> Building dependency tree
> Reading state information... Done
> Package libavformat56 is not available, but is referred to by another package.
> This may mean that the package is missing, has been obsoleted, or
> is only available from another source
> 
> E: Package 'libavformat56' has no installation candidate
> 

For all of the above packages, I am able to install a higher version, or the -dev version, however when it comes to libmysqlclient18, I get the same kind of error as the previous 2 packages, but when searching for a different version, there are no packages available starting with libmysqlclient at all, so I'm not sure where to go from here to keep trying to get this working.

I"ll be very grateful for any help anyone can give, thanks.

nabh, 2017-08-24:
Same problem here! Hoping for a fix/workaround

DawsonLeBrown, 2017-08-24:
For now, I've just set it up with the previous raspbian lite image - available here https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2017-07-05/ This is the last jessie image before the current stretch image was added. With this image, following the instructions to set up motioneye is a breeze.

kevinmcaleer, 2017-08-25:
Same issue

barneyc, 2017-09-01:
Arghhh

So far, to get around this particular issue.. use the package installer in Raspbian (I used the GUI) and install Motion from there.  It will handle these dependencies.

After that you will probably run into issues at `pip install motioneye` with a script error.  This needs a manual updating of pycurl I think

wb666greene, 2017-09-22:
I think with these instructions, this issue can be closed:

I set my hostname to picam after installation and changing the default password.
This is how I then installed Motioneye on a Pi2 with Raspbian-Stretch (2017-09-07):

// steps on PiZeroW, Pi2:
sudo apt-get update
sudo apt-get dist-upgrade
//
sudo nano /etc/modules
// add:
bcm2835-v4l2
//
//
// extra packages:
sudo apt-get install libssl-dev libcurl4-openssl-dev libmariadbclient18 libpq5 mysql-common ffmpeg
//
// prebuilt motion deb package:
wget https://github.com/Motion-Project/motion/releases/download/release-4.0.1/pi_stretch_motion_4.0.1-1_armhf.deb
sudo dpkg -i pi_stretch_motion_4.0.1-1_armhf.deb
//
// install motioneye:
sudo pip install motioneye
//
// setup to run motioneye:
sudo mkdir -p /etc/motioneye
sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
sudo mkdir -p /var/lib/motioneye
sudo cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
sudo systemctl daemon-reload
sudo systemctl enable motioneye
sudo systemctl start motioneye
//
// To enable web control from local network, seems missing from the docs:
// edit both /etc/motioneye/motioneye.conf and /etc/motioneye/motion.conf and then reboot.
// In /etc/motioneye/motion.conf Change webcontrol_localhost on To webcontrol_localhost off
// In /etc/motioneye/motioneye.conf Change motion_control_localhost true To motion_control_localhost false
// then reboot system
// can pause and resume motion detection from a terminal or node-red etc. with commands:
curl http://picam:7999/1/detection/pause -s -o /dev/null
curl http://picam:7999/1/detection/start -s -o /dev/null

tigrrr, 2017-09-23:
many thanks for the guide! I have a question, in order to enable samba share from motioneye, motionEye is required to run as root according to motioneye.conf, is this possible to enable or would it be better to define a share manually ?

bekato, 2017-11-13:
many thanks for the guide!!!
sorry for my bad english.
I follow the instruktions without default
Bat for this instruction : 
 // In /etc/motioneye/motion.conf Change webcontrol_localhost on To webcontrol_localhost off
I don't have this file:  "/etc/motioneye/motion.conf"

bekato, 2017-11-13:
PS:
I find the file in /etc/motion/motion.conf

kim3er, 2017-12-03:
Thanks for the guide @wb666greene, it really helped. Only two packages are missing from you guide, starting from a fresh installation, `libjpeg-dev` and  `python-pip`.

I've written up the guide as a commit, so that it can hopefully be included in the wiki. @ccrisan, would it be possible to update the wiki, as per the commit below?

https://github.com/kim3er/motioneyewiki/commit/b2a298020ec8157920327ec1dbf35f53ce7f5e85

ccrisan, 2017-12-03:
@kim3er actually I have rewritten the Install On Raspbian page so that it reflects Raspbian Stretch. Things are a lot simpler these days (we have ffmpeg back in Debian) so instructions are now straight forward.

I'll close the issue as things go smooth with a fresh, up-to-date Raspbian Stretch setup, here. If you still encounter issues, please feel free to comment here and I'll look into it.

closed 2017-12-03 by ccrisan

AbbasKagdi, 2018-01-27:
tried everything, still have problems.
Problems in dpkg, pip, and motion.  **:(**

its-complete, 2018-02-07:
hi all, @wb666greene  I'm on the "sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf" (also step 6 [here](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian))

I am getting "no such file or directory." Any ideas how to troubleshoot?

ccrisan, 2018-02-08:
@its-complete sometimes stuff gets installed in `/usr/share/motioneye` instead of `/usr/local/share/motioneye`, depending on your system/python/pip installation. Therefore you could try leaving out `/local` from your command.

okoloboha, 2019-04-17:
Tray this  /home/username/.local/share/motioneye/extra/

mooniker, 2019-07-09:
I think I found that the pillow installation error interrupts the motioneye installation (on buster/stable). I ran `apt-get install python-pillow` after the error and than reran `pip install motioneye` and everything is where it should be whereas before re-pipping, the conf/usr files were no where to be found.