motioneye-project/motioneye #637: Video on "wired PC" but not working on wireless

lodgemoorlad, 2017-08-23:
Hi, hope someone can help.
I can see the action from motioneye on my "hardwired PC" but can't see it when I'm using my home wifi on my smartphone, it has worked well in the pass

Also I'm NOT using the wifi onboard motioneye, I use a powerplug so that I can get the range/distance involved.
I also could'nt get it to stream by putting :8081 at the end of motioneye's ip address, 
Hope someone can sort it for me as I've run out if ideas !

chrisf4lc0n, 2017-08-23:
Can you recollect what has changed in your network since "it has worked well in the past"?

lodgemoorlad, 2017-08-23:
Hi
Thanks very much for getting back to me

It suddenly wasn't working one day when I went to look at it, but it was
several weeks before that since I checked it out

However I have had to change my router and I have changed my phone , but
nothing changed, I could see the action on my wires PC, BUT nothing on the
house WiFi
A

chrisf4lc0n, 2017-08-24:
Blocked ports for WiFi devices? Check your firewall setting on your router.

lodgemoorlad, 2017-08-24:
Hi Krzysztof

Thanks for your help
I tried the firewall and it appeared to work!

The BUT!
When I did that I was using my iPad. But as soon as I tried my android
smartphone, it didn't work
This is how it was before, but it also didn't work on my iPad

So I checked all my devices, it's now working on two iPads and one tablet
It's not working on two android phones I have , how very very strange is
that, infact if someone said this to me, I wouldn't believe them
I've never experienced this type of thing ever
A

chrisf4lc0n, 2017-08-24:
The problem definitely lies on the router side. You may have some sort of access policies set up. Without knowing what router you have got it would be difficult to help any further.

JamBalaya56562, 2026-11-07:
Closing as answered: as Krzysztof found above, this was on the router side. After the firewall change the stream worked on the iPads and the tablet, and the Android phones that still could not reach it point to an access policy on the router, not to motionEye.

closed 2026-11-07 by JamBalaya56562