motioneye-project/motioneye #641: bad image

alfredovilla, 2017-08-26:
Hi, i have configured my raspberry (Raspbian 8.0) with motionEye (0.36.1) and Motion (4.0.1). 
Add my ip camera (rtsp://192.168.1.134:554/onvif2) correctly but the result streaming is a green bad imag (attached files). If i adding the same camera streaming to VLC the image streaming is very good.. what's the problem?

thnx
[image]

Harvie, 2017-09-29:
What version of ffmpeg you have? Try different versions.

JamBalaya56562, 2026-11-07:
Closing, as the question about the ffmpeg version went unanswered, and this was with Motion 4.0.1 and motionEye 0.36.1. Decoding the RTSP stream is done by Motion, so if a camera still gives a green picture with a current Motion while VLC plays the same URL fine, please open a new issue with `motion.log`, or report it to [Motion](https://github.com/Motion-Project/motion).

closed 2026-11-07 by JamBalaya56562