motioneye-project/motioneye #643: Cannot Install Motion on Raspian

tomfcoughlan, 2017-08-29:
I have tried with 3 different sd cards and updated and upgraded Raspian but am consistently getting the same messages:

root@raspberrypi:/home/pi# apt-get install python-pip python-dev curl libssl-dev libcurl4-openssl-dev libjpeg-dev libx264-142 libavcodec56 libavformat56 libmysqlclient18 libswscale3 libpq5
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package libmysqlclient18 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libavcodec56 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

Package libavformat56 is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Unable to locate package libx264-142
E: Package 'libavcodec56' has no installation candidate
E: Package 'libavformat56' has no installation candidate
E: Package 'libmysqlclient18' has no installation candidate
E: Unable to locate package libswscale3

vsidlu, 2017-08-31:
it seems you are using new verison of raspbian - strech. on previous version jessie everything is installing and working smoothly. there is place you can download old versions - https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2017-07-05/

barneyc, 2017-09-01:
Downloaded the full non-stretch Jessie (June or July I think) and the same problem exists.  All of packages are a version higher :o(  No idea how to resolve that

rdmorin, 2017-09-03:
I was able to get this working by adding the two previous releases to my sources.list file. It's full contents are: 
deb http://mirrordirector.raspbian.org/raspbian/ stretch main contrib non-free rpi
deb http://mirrordirector.raspbian.org/raspbian/ jessie main contrib non-free rpi
deb http://mirrordirector.raspbian.org/raspbian/ wheezy main contrib non-free rpi

After making that change, run "apt-get update" and try the installation again.

Firesphere, 2017-09-04:
You'll need to use the newer packages for Stretch, they are:
- libx264-148
- libavcodec57
- libavformat57
- default-libmysqlclient-dev
- libswscale4

I did not need libs. After that, installing Motion from the repository should work fine, and you should be able to continue as per the docs.

Kywalh, 2017-09-13:
@Firesphere 
I tried to install the newer package but get still issues...

`Dépaquetage de motion (4.0.1-1) ...
dpkg: des problèmes de dépendances empêchent la configuration de motion :
 motion dépend de libavcodec56 (>= 6:11~beta1) | libavcodec-extra-56 (>= 6:11.7) ; cependant :
  Le paquet libavcodec56 n'est pas installé.
  Le paquet libavcodec-extra-56 n'est pas installé.
 motion dépend de libavformat56 (>= 6:11~beta1) ; cependant :
  Le paquet libavformat56 n'est pas installé.
 motion dépend de libavutil54 (>= 6:11~beta1) ; cependant :
  Le paquet libavutil54 n'est pas installé.
 motion dépend de libmysqlclient18 (>= 5.5.24+dfsg-1) ; cependant :
  Le paquet libmysqlclient18 n'est pas installé.
 motion dépend de libswscale3 (>= 6:11~beta1) ; cependant :
  Le paquet libswscale3 n'est pas installé.

dpkg: erreur de traitement du paquet motion (--install) :
 problèmes de dépendances - laissé non configuré
Traitement des actions différées (« triggers ») pour systemd (232-25+deb9u1) ...
Traitement des actions différées (« triggers ») pour man-db (2.7.6.1-2) ...
Des erreurs ont été rencontrées pendant l'exécution :
 motion`

wb666greene, 2017-09-21:
I installed Motioneye on Raspbian-stretch PiZero-W, everything seems to work, except I can't connect to the control port (7999) even though I changed motion_control_localhost from true to false and rebooted.  The control port does run from Chromium  locally on the PiZero-W using  localhost:7999 (but slow as all get out).  Remote admin and monitoring on port 8765 seems to work correctly.

The details of what I did and how I overcame the errors encountered installing on Stretch are here:
https://groups.google.com/forum/#!topic/motioneye/e-RoEWQvMuI

I plan on repeating it with Stretch on a Pi2 with WiPi USB WiFi dongle,  as there appears to be a problem with the WiFi firmware for the chip used on the PiZero-W (and Pi3) that makes them unsuitable for 24/7 usage.  Details are here:
https://github.com/raspberrypi/linux/issues/1342

Once I've completed this, I'll post a new topic to Google groups with hopefully commands that you can cut and paste to install.

wb666greene, 2017-09-22:
I set my hostname to picam after installation and changing the default password.
This is how I then installed Motioneye on a Pi2 with Raspbian-Stretch:

// steps on PiZeroW, Pi2:
sudo nano /etc/modules
// add:
bcm2835-v4l2
//
//
// extra packages:
sudo apt-get install libssl-dev libcurl4-openssl-dev libmariadbclient18 libpq5 mysql-common ffmpeg
//
// prebuilt motion deb package:
wget https://github.com/Motion-Project/motion/releases/download/release-4.0.1/pi_stretch_motion_4.0.1-1_armhf.deb
sudo dpkg -i pi_stretch_motion_4.0.1-1_armhf.deb
//
// install motioneye:
sudo pip install motioneye
//
// setup to run motioneye:
sudo mkdir -p /etc/motioneye
sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
sudo mkdir -p /var/lib/motioneye
sudo cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service
sudo systemctl daemon-reload
sudo systemctl enable motioneye
sudo systemctl start motioneye
//
// To enable web control from local network, seems missing from the docs:
// edit both /etc/motioneye/motioneye.conf and /etc/motioneye/motion.conf and then reboot.
// In  /etc/motioneye/motion.conf  Change  webcontrol_localhost on  To  webcontrol_localhost off
// In /etc/motioneye/motioneye.conf Change  motion_control_localhost true  To  motion_control_localhost false
// then reboot system
// can pause and resume motion detection from a terminal or node-red etc. with commands:
curl http://picam:7999/1/detection/pause -s -o /dev/null
curl http://picam:7999/1/detection/start -s -o /dev/null

timrowledge, 2017-12-04:
Unfortunately none of the above seems to work on a fresh 20170907 Stretch on a Pi 3.
The original instructions regarding dependencies (and aren't they supposed to sort out automatically?) resulted in the same problems that tomfcoughlan explained. I then tried wb666greene's package list, which at least all loaded.
But the pip install motioneye only produces 
```
Collecting motioneye
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 353, in run
    wb.build(autobuilding=True)
  File "/usr/lib/python2.7/dist-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/usr/lib/python2.7/dist-packages/pip/req/req_set.py", line 554, in _prepare_file
    require_hashes
  File "/usr/lib/python2.7/dist-packages/pip/req/req_install.py", line 278, in populate_link
    self.link = finder.find_requirement(self, upgrade)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 465, in find_requirement
    all_candidates = self.find_all_candidates(req.name)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 423, in find_all_candidates
    for page in self._get_pages(url_locations, project_name):
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 568, in _get_pages
    page = self._get_page(location)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 683, in _get_page
    return HTMLPage.get_page(link, session=self.session)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 792, in get_page
    "Cache-Control": "max-age=600",
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 501, in get
    return self.request('GET', url, **kwargs)
  File "/usr/lib/python2.7/dist-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 488, in request
    resp = self.send(prep, **send_kwargs)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/sessions.py", line 609, in send
    r = adapter.send(request, **kwargs)
  File "/usr/share/python-wheels/CacheControl-0.11.7-py2.py3-none-any.whl/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/usr/share/python-wheels/requests-2.12.4-py2.py3-none-any.whl/requests/adapters.py", line 423, in send
    timeout=timeout
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/connectionpool.py", line 643, in urlopen
    _stacktrace=sys.exc_info()[2])
  File "/usr/share/python-wheels/urllib3-1.19.1-py2.py3-none-any.whl/urllib3/util/retry.py", line 315, in increment
    total -= 1
TypeError: unsupported operand type(s) for -=: 'Retry' and 'int'
```
I even tried the suggestions on http://www.howtoembed.com/projects/raspberry-pi/95-motioneye-with-raspberry-pi where something called python-tornado is added. Same results.

ccrisan, 2017-12-04:
Latest raspbian to date is 2017-11-29. Please try using that and see if your problem persists.

timrowledge, 2017-12-04:
Yup, working on that as I type. Let's hope.

timrowledge, 2017-12-04:
OK, after a false start because I forgot to apt-get update and got wrong versions of assorted stuff, it all worked just nicely with your instruction as of 3 Dec 2017. Excellent!

Now all I have to do is understand what I'm doing...

ccrisan, 2017-12-04:
The issue can then be closed.

closed 2017-12-04 by ccrisan