motioneye-project/motioneye #645: Errors in log

jasonlshelton, 2017-08-31:
Motioneye starts and runs fine for a while, then I will randomly get an email that my disk is running out of space, so I check and my syslog is huge. tail -f shows me the following message repeating extremely rapidly

```
Aug 31 09:38:23 Doc meyectl[365]: INFO: interrupt signal received, shutting down...
Aug 31 09:38:23 Doc meyectl[365]: ERROR: Exception in callback None
Aug 31 09:38:23 Doc meyectl[365]: Traceback (most recent call last):
Aug 31 09:38:23 Doc meyectl[365]: File "/usr/local/lib/python2.7/dist-packages/tornado/ioloop.py", line 887, in start
Aug 31 09:38:23 Doc meyectl[365]: fd_obj, handler_func = self._handlers[fd]
Aug 31 09:38:23 Doc meyectl[365]: KeyError: 17
```

Stopping motioneye stops the errors. Restarting motioneye does not start the messages again, but at some point they return and it all starts over.

Any idea what's going on? or what more info do you need?

Raspbery pi2 running raspian, using a MS USB camera

ccrisan, 2017-09-01:
Please **attach** the entire log file.

jasonlshelton, 2017-09-01:
The entire syslog file was 2.1Gb. I deleted it to recover the space. the next time it does this, I'll post the top of it before the millions of entries like above.

jasonlshelton, 2017-09-06:
happened again, the logs are 3.9gb Would you still like me to post them?

`Sep  6 09:02:01 Doc meyectl[26110]: Traceback (most recent call last):
Sep  6 09:02:01 Doc meyectl[26110]: File "/usr/local/lib/python2.7/dist-packages/tornado/ioloop.py", line 887, in start
Sep  6 09:02:01 Doc meyectl[26110]: fd_obj, handler_func = self._handlers[fd]
Sep  6 09:02:01 Doc meyectl[26110]: KeyError: 20
Sep  6 09:02:01 Doc meyectl[26110]: ERROR: Exception in callback None`

ccrisan, 2017-09-06:
This doesn't make too much sense as an exception. It looks as though somehow the Python interpreter's memory gets corrupted. I would like to see, nevertheless, a larger excerpt from your log file. Could you `tail -n 10000` from the log file and **attach** it here?

Have you overclocked your system? If yes, you might want to try this with overclocking disabled. Another idea that may not be applicable, but is worth mentioning here would be to try to replace the unit, just to rule out any faulty hardware.

jasonlshelton, 2017-09-07:
If I did that, it would just be the lines I pasted already, repeating over and over.

What I have figured out is that it happens when I try to view the saved pictures and the listing takes too long, if I close the browser before the error comes back, it seems to trigger this race condition. It's not consistent, so there must be some other variable. For now, I just adjusted the picture settings to make the directory listings go faster and hopefully avoid this altogether

ccrisan, 2017-09-07:
@jasonlshelton nevertheless, the error doesn't make sense. Many users keep thousands of media files and motionEye has a hard time parsing them all, but this error... I have never seen it.

JamBalaya56562, 2026-11-07:
Closing, as the cause was never found, and this was the Python 2 release with an old Tornado (`/usr/local/lib/python2.7/dist-packages/tornado/ioloop.py`). Current releases run on Python 3 with a current Tornado. If the log fills with a repeating error again with a current release, please open a new issue with a larger excerpt of the log.

closed 2026-11-07 by JamBalaya56562