motioneye-project/motioneye #646: Backyard bird activity

dabi236, 2017-09-01:
I have setup a Raspberry Pi camera in my backyard to view bird house activity.  The hardware/software works great but when it gets to dark for the solar panel and eventually the battery power gets to low I'm unable to log into the Pi (this is what I would expect).  When the sun starts charging the battery again I'm unable to log into the Pi to view live video.  When I disconnect the power momentarily from the Pi and let it do a reboot I can once again see video.  I suspect that the Pi doesn't actually shut off when the battery power gets to low but sheds the wifi capability and maybe some other power hungry sub systems.  When the power is restored from the solar panel (without a reboot) the sub systems that were shed will not actually reset until power is cycled.  If this is indeed what is happening is there anyway to force the Pi off and to stay off when this threshold voltage is reached?

My hardware is listed below:

-Raspberry Pi Zero W with NoIR camera
-Colossal 6V 9W Solar Panel
-USB/DC/Solar Lithium Ion/Polymer Charger
-DROK Mini USB DC-DC 3V to 5V Voltage Step Up Converter Volt qBoost Regulator Module
-Lithium Ion Battery Pack - 3.7V 6600mAh

Firesphere, 2017-09-07:
What you could do, is run an hourly cronjob to restart the systems that are shutdown, based on the power supply.

e.g. (non-functional, but a gist of what I mean):
```
#!/bin/not-actually-bash
VOLTS = vcgencmd measure_volts core
TIME = date +"%H"
if (VOLTS gte 900 && TIME gte 9) then
    if (check if services are not running here) then
        service {servicename} restart
   fi
fi
```

Harvie, 2017-09-29:
Just use some brown-out prevention circuitry that will completely disconnect your raspi once power drops under 5V and reconnects it back as it reaches 5V again. It will work even better if you put it between battery and 5V step down you use to power raspi (i expect that your solar battery system runs higher than 5V, eg. if you have 12V lead acid battery, you can shutdown raspi as it drops below 10.5V and reconnect it as it reaches 11.5V or something similar). Google "low voltage disconnect".

JamBalaya56562, 2026-11-09:
This is about the power supply rather than motionEye; Harvie's low voltage disconnect is the usual way to get a clean power cycle. Closing.

closed 2026-11-09 by JamBalaya56562