motioneye-project/motioneye #649: Hard refresh needed when changing some settings (0.37)

sfldcr, 2017-09-04:
This is a comment, more than a question. I'm not sure if this is observed behavior from others, but I noticed it today.

I have two CHiPs that are independently running MotionEye (connecting them to feed a central system is a future step). I updated both to 0.37 today, and I was also updating and unifying some of the settings so they were more consistent across both cameras. I noticed that after I changed the settings once or twice, on each system (separately), the UI no longer allowed me to make settings changes. (Error: "An error occurred. Refreshing is recommended.") This hasn't happened to me before, and I don't think I was doing anything abnormal this time that would have caused this. After several CHiP reboots, checking a variety of things, and then searching through previous tickets, I saw that a hard browser refresh sometimes solves this, and it appears to have solved it for me this time.

I just thought I would mention this - could there have been a change in 0.37 that is causing this? Or was it just a dumb user (me) error? Either way - things appear to be working now, so I just wanted to mention this, and tell people what has worked for me.

ccrisan, 2017-09-04:
@sfldcr can you please clear your browser's cache, refresh and try again?

sfldcr, 2017-09-04:
Sorry, I should have said this better - I already solved the issue (and a hard refresh worked). I just wanted to comment that it happened to me on 0.37, but never happend before. Not sure if that is a coincidence, or if there was a change in 0.37 that could have made this more likely to happen. Thanks for the quick comment, and for your work on this!

ccrisan, 2017-09-05:
There have been some authentication related changes in 0.37 so yes, that might have caused it.

closed 2017-09-05 by ccrisan