motioneye-project/motioneye #650: Random grey areas in camera output

lastfreenick, 2017-09-05:
I've always had some random grey areas popping up in one of my camera's video output (usually starting at the bottom of the picture,)and thought it would be either caused by the poor wifi connection or the use of an old model b pi board (too much work for this slower model?). But recently, I've upgraded both the pi to newer model 2 and significantly improved wifi. 

Still I get these grey areas a lot, I feel even more now. Is this just my camera hardware failing? It's a noir model camera by the way.

lastfreenick, 2017-09-05:
Here is one extreme example
[image]

ccrisan, 2017-09-05:
Are you using motion 4.x? If not, upgrade. If yes, please report this to [motion](https://github.com/Motion-Project/motion).

closed 2017-09-05 by ccrisan