motioneye-project/motioneye #651: As of latest update, port 8081 doesn't accept the username/password anymore

Firesphere, 2017-09-05:
I don't exactly know when it started, but the Basic authentication seems to have been broken. I am unable to access it anymore.

ccrisan, 2017-09-05:
Please give us more details:
1. What motionEye version?
2. What motion version?
3. What hardware?
4. What OS?

If I understood correctly, you have one local camera (USB/CSI) and you have enabled Basic Authentication in Settings Panel > Video Streaming. Now when you access your system's IP on port 8081, it asks for credentials and they no longer work? Did I understand this correctly?

Firesphere, 2017-09-05:
Sorry, it was a bit late :)

- MotionEye 0.37
- Raspberry Pi B 3
- Motion 4.0-1 (Latest Stretch version https://packages.debian.org/stretch/motion)
- Raspbian Stretch

I have a CSI camera connected and the dashboard on default port (8765) works fine. The access on port 8081 however, requires Basic Authentication.
Although I do get the dialog, asking for authentication, even with the correct username and password, I keep getting the authentication dialog and never actually reach the stream.

ccrisan, 2017-09-06:
@Firesphere it seems there's a problem indeed with the authentication. I'll look into it and keep you posted.

Firesphere, 2017-09-07:
Thanks @ccrisan 
Sorry again for the terrible initial issue report, it was late and I didn't want to forget to make an issue

ccrisan, 2017-09-07:
7a8b5d78bee059f98bedd69d5cb3416a44d64c81 fixes this issue. It will be part of the 0.37.1 bug fix release.

closed 2017-09-07 by ccrisan

Firesphere, 2017-09-17:
I still seem to have this issue, even after updating. A reboot after updating did not solve anything either.

No changes except now on MotionEye 0.37.1.

ccrisan, 2017-09-17:
Have you cleared your browser cache and have you updated your normal (surveillance) password?

Firesphere, 2017-09-17:
I'm experiencing the same issue, on a machine that I have not used to access MotionEye before. This laptop has literally been gathering dust for at least 3 months, and it has the same issue in all browsers.

Just to be sure, I cleared all browser caches. Same issue occurs :(
Except for Safari, but Safari just now simply even refuses to load (most probably a Safari issue, as I have the same on my old iPhone and another MacBook from my neighbours I tried)

Firesphere, 2017-09-17:
I'll try again on my office machine tomorrow, as I don't have access to it at the moment, Clear cache, retry access (I have a static IP and forward set up correctly, I do get the authentication dialog)

ccrisan, 2017-09-17:
I'm asking you again: have you updated your normal (surveillance) password (since upgrading to 0.37.1)?

Firesphere, 2017-09-17:
Ow, sorry, missed that in your question, but yes, I have updated the password immediately after updating, because I figured I needed to do that.

I've just done that again now, but the issue persists :(

ccrisan, 2017-09-17:
@Firesphere so just to be sure:
* you are running motionEye 0.37.1
* you have updated your surveillance (normal) password since upgrading to 0.37.1
* your surveillance user is still called "user"
* you have cleared your cache
* you have selected Basic in Video Streaming > Authentication Mode
* when you access your motionEye's IP address on port 8081, on Chrome, it keeps asking for credentials, even though you're typing in the correct ones ("user" and whatever you chose for surveillance password).

Are all the above correct?

Firesphere, 2017-09-18:
Hi @ccrisan 
I'll go point by point :)
- Yes, I'm running MotionEye 0.37.1
- I have updated the normal password since with the normal username `user`
- I have attempted to change the username to `surveillance` and back
- I have cleared my cache and tried using private mode
- Yes, I have Basic as authentication mode
- This happens not just in Chrome, but in all browsers (Firefox, Chromium, Edge and Internet Explorer, in my case)

A thing that might be of issue though, is that I initially (before updating MotionEye), changed the username. Might that be the origin of the persisting issue?

ccrisan, 2017-09-18:
Can you please check your `thread-1.conf` (usually found at `/etc/motioneye`) and look for the following lines:
* the one starting with `# @normal_username`
* the one starting with `# @normal_password`
* the one starting with `stream_authentication`

The username and passwords should be in clear text and should correspond. If that's not the case, I would like to see that file (you could replace your passwords with some **** where (if) they appear in the file.

Firesphere, 2017-09-18:
This is my thread file, the password is _not_ in clear text
```
# @webcam_resolution 50
# @upload_subfolders on
# @upload_server
# @enabled on
# @network_server
# @upload_username
# @motion_detection on
# @upload_port
# @upload_location
# @preserve_movies 1
# @network_username
# @upload_movie on
# @id 1
# @upload_password
# @upload_method post
# @upload_picture on
# @working_schedule_type outside
# @network_password
# @upload_service ftp
# @name RPi 1
# @preserve_pictures 1
# @storage_device custom-path
# @network_share_name
# @upload_enabled off
# @webcam_server_resize on
# @working_schedule
# @manual_record off
# @manual_snapshots on

ffmpeg_output_movies on
height 960
stream_quality 50
threshold 27646
quality 100
noise_level 31
stream_auth_method 1
stream_motion 1
target_dir /var/lib/motioneye/Camera1
stream_authentication user:my_encrypted_password
mask_file
width 1440
```
Manually updating the password to plaintext however, resolved the issue! I've taken out the sensitive data. If you want, I can email or message them :)

I'm roughly guessing, the issue is if the usernames don't match the defaults, the update isn't written? (But that's just my thought :) )

ccrisan, 2017-09-18:
My apologies, you'll have to look into `motion.conf` as well for the normal_username/password lines.

Firesphere, 2017-09-18:
No problem :)

`# @normal_username surveillance`
`# @normal_password **encrypted_string**`

ccrisan, 2017-09-18:
@Firesphere are you 100% sure you're running motionEye 0.37.1? What's the version that it reports in the UI? I'm sorry to ask this but the encrypted normal password doesn't make sense given commit https://github.com/ccrisan/motioneye/commit/7a8b5d78bee059f98bedd69d5cb3416a44d64c81?

Firesphere, 2017-09-18:
You are a genius.

Yes, I have upgraded and run the latest MotionEye upgrade commands. However, I'm running a custom installation, which includes a set of custom sensor readings etc. on a base Raspbian. It seems a manual update _failed_ instead of succeeded (without notifying me). I will try to update manually again and probably stop bothering you. Thanks for your help!

ccrisan, 2017-09-18:
I'm glad you sorted it out.

Firesphere, 2017-09-18:
I'm sorry for the confusion and time I cost you!