motioneye-project/motioneye #654: MotionEye saves images and mp4 but can't retrieve them

enterprisemike, 2017-09-09:
Hi,

I'd really appreciate some guidance on this one. I have MotionEye setup with an external USB drive mounted at /data/media. Image and video is being written there with no problem. I can see them via samba and SSH.
However on the MotionEye web interface, when I try to view images or video I get:
"Pictures taken by [Camera name]"
(no media files)

The File Storage settings for the cameras are:
Storage Device: WD My (/dev/sda1)
Root Directory: /footage/[CameraName]

It seems like MotionEye is looking somewhere else for the images. If I change the setting to write back to the SD card, I still can't retrieve the images through the interface.

Can someone point me in the right direction to work out what's going on?

Thanks
Mike

ccrisan, 2017-09-09:
First off, given the `/data/media` path, I strongly believe you're using motionEyeOS and this issue would belong to [motionEyeOS](https://github.com/ccrisan/motioneyeos) instead of here.

In any case, motionEye is indeed looking in the root directory and searches for files recursively. If your media files are in `/data/media/sda1/footage/[Camera Name]` (which is what I assume given the details you offered here), then your Storage Device should be `/dev/sda1`  and your Root Directory should be `/footage/[Camera Name]`. So please stick with the first variant of your settings and check if the files are indeed on that path.

enterprisemike, 2017-11-04:
Thank you for that info, I'll transfer the issue to motionEyeOS.

closed 2017-11-04 by enterprisemike