motioneye-project/motioneye #656: Picture file upload to server

feh123, 2017-09-10:
I have updated to 20170827 and I cannot get the file upload to work anymore. I am using a RPI3 with a c170 logitech webcam. I set the target as FTP server - to use a WDMyCloud NAS on a reserved IP address. I have double checked the IP address, port, password for the NAS - they all seem fine. The source directory is custom path (/data/output/camera1). I have subfolders enabled. When I hit test service it takes 4-5 minutes and then I get the message "error occurred refresh recommended". The motion detection/file storage on the custom path is okay and I can open the files and view them on the RPI. No file transfer to the NAS occurs. As anyone seen this as well?

  
[motioneye.txt](https://github.com/ccrisan/motioneye/files/1290741/motioneye.txt)

chrisf4lc0n, 2017-09-10:
Is this MotionEye or MotionEyeOS?

feh123, 2017-09-10:
Hi, this is MotionEyeOS.

ccrisan, 2017-09-11:
Please attach your motioneye.log file.

feh123, 2017-09-11:
Hi, log file attached - I renamed it txt to get it accepted. Yours Fr
[motioneye.txt](https://github.com/ccrisan/motioneye/files/1294168/motioneye.txt)
ed

ccrisan, 2017-09-12:
That `EOFError` that you get whenever accessing the ftp server indicates some problem communicating to the server.
1. Does it work with the good old `ftp` command? Or does it work with some other UI tool such as Filezilla?
2. Can I see a screenshot of your ftp settings in motionEye?

feh123, 2017-09-12:
Hi screen shot:
[image]

I am not very good with command line. I have remote terminal on my laptop and can SSH into the motioneye - what do you want me to do re ftp? Thanks Fred

ccrisan, 2017-09-12:
Your location doesn't look good. It looks more like a SMB path rather than an FTP one - make sure you use a /path/to/location format and that the actual path exists on the FTP server.

You don't have to SSH into motionEyeOS to test this. You could use Filezilla from your laptop, while it's connected to the same network as your motionEyeOS board.

feh123, 2017-09-12:
Hi I used filezilla and checked the ftp path. It was different and on changing to the filezilla path the upload test worked! Thanks! How often does the upload occur? Can I control the timing of the upload? Now I just need to get port forwarding to work on my router! Currently I cannot see the motion files from the internet as I cannot get an external port to work. Could I add a port to motioneye that I could see? Thanks for all your help!

ccrisan, 2017-09-12:
The upload happens each time a media file is created. It actually happens when a picture is taken or after a movie ends.

You can do port forwarding and that has nothing to do with motionEye. See https://github.com/ccrisan/motioneyeos/wiki/FAQ#what-ports-do-i-have-to-open-in-my-firewall and https://github.com/ccrisan/motioneyeos/wiki/FAQ#can-i-access-motioneyeos-from-the-internet.

closed 2017-09-12 by ccrisan