motioneye-project/motioneye #657: Allow user to override hostname

ioquatix, 2017-09-11:
My server has several hostnames and the default one is not the one I want for this install.

I manually edited this line: https://github.com/ccrisan/motioneye/blob/306155120776f36140794a1373df8b25ae639692/motioneye/handlers.py#L244

I feel like we should make this a config option. I'd be happy to make a PR if that was acceptable, with the default being the current behaviour unless the configuration is specified.

What do you think?

ccrisan, 2017-09-11:
I I don't see the point in making the hostname configurable. It doesn't have too much importance in motionEye's functioning.

ioquatix, 2017-09-11:
I put motion eye on a public IP behind nginx and I don't want it revealing the host name of the server.

ccrisan, 2017-09-11:
Alright if you can provide a PR to make this configurable while providing backwards compatibility, I'll review and accept it.

Dycast, 2017-09-13:
I'm running this on my local server and allow a remote desktop approach to reach it when not home, it would help greatly if the ports (ie: :80 ) were also configurable, I believe the seperate cameras are, but not the main one, which I like to use as I can see all cameras on one screen. - Currently I'm running the OS ver, but am looking at using this one laid on top of a distro as I'm expanding and have several RPI's running this around the house, several of which I want to have a 'local monitor' screen attached

ioquatix, 2017-12-10:
@ccrisan Can you give me a very rough idea of how you'd like this implemented?

ccrisan, 2017-12-10:
@ioquatix a "@server_name" configuration option would be the best way to implement it. It can be empty (missing) by default in which case the hostname will be used.

JamBalaya56562, 2026-10-25:
This has been implemented by @hobbyquaker in #846 (merged in 2018) as the `server_name` option, following the approach suggested above:
add `server_name <name>` to motioneye.conf, and it is shown in the page title and header instead of the system hostname.
If unset, the hostname is used as before.

Closing as completed.
Feel free to reopen if anything is still missing.

closed 2026-10-25 by JamBalaya56562