motioneye-project/motioneye #658: email interval snapshots

arrfour, 2017-09-11:
Anyone know a way to email snapshots on a regular interval (say 30 minutes) regardless of motion detection? I have a high activity area I just want to send a scheduled email to show the content level of a container to an outside party. Can't give them access to the motionEye server on our network, but I can send them emails every 30 minutes or so.

I'm banging my head against the wall on this and Google is failing me. 

Have all other aspects of motionEye working properly.

discarn8, 2017-09-11:
Via *nix or Windows?

If windows... Sorry, I'd be of no help

If *nix... Check this thread out, for the capture part.  
https://superuser.com/questions/663928/ffmpeg-to-capture-stills-from-h-264-stream

As for the email portion, I'd say a simple smtp cron job would be sufficient

arrfour, 2017-09-11:
I'm hoping to do it all within the motionEye interface. I'm also looking at mutt from the linux host

chrisf4lc0n, 2017-09-11:
If you are running MotionEye you can configure ssmtp and then create bash script something along them lines:
```
#!/bin/sh
d=`date +%d-%m-%y`
t=`date +%T`
curl -o /tmp/$t$d.jpg 'http://'YOUR MOTIONEYE IP':8765/picture/'NUMBER OF YOUR CAMERA'/current/?_username=admin&_signature=WHATEVER IT IS FROM THE STREAM URL'
mpack -s "WHATEVER THE SUBJECT YOU WANT" /tmp/$t$d.jpg YOUR EMAIL
rm /tmp/$t$d.jpg
```

arrfour, 2017-09-12:
curl is an excellent idea! I've been able to use MUTT for the email piece in the past. I'll try this. It would be great to have this baked into MotionEye as well, preferably on a per camera setting.

arrfour, 2017-09-12:
Based on chrisf4lc0n and discarn8 suggestions I slapped together the following as a script. Thanks for the help everybody. I wish I was capable of adding this to the motionEye interface.

camtest.sh #This script can be set to run as a cron job.

#!/bin/bash
#Define Addresses
a="emailaddress@domain.com"
bcc="blindcopyaddress@domain.com"

#Pull images with curl
echo "--------Pulling images"
#Pull the current  image via curl
curl -o /tmp/cam4.jpg 'http://<ip-address>:8765/picture/4/current/'
#Pull the current  image via curl
curl -o /tmp/cam3.jpg 'http://<ip-address>:8765/picture/3/current/'
#Pull the current image via curl
curl -o /tmp/cam5.jpg 'http://<ip-address>:8765/picture/5/current/'

#create email with all attached temp shots and send via mutt
echo "--------Sending email"
mutt -s "<subject text>" -b $bcc -a /tmp/cam*.jpg < email.txt -- $a

#clean up tmp folder
echo "--------Cleaning up"
rm /tmp/cam*.jpg

#finish
echo "--------Completed"

closed 2017-09-12 by arrfour

chrisf4lc0n, 2017-09-12:
> mutt -s "" -b $bcc -a /tmp/cam*.jpg < email.txt -- $a

You need the key after /current.

chrisf4lc0n, 2017-09-12:
Otherwise all good!
Great job!

arrfour, 2017-09-12:
My install does not list a key at the end of the snapshot url for each camera so this script worked for me. Others might need to tweak like chrisf4lc0n points out.