motioneye-project/motioneye #659: Notification on / off via Command line? Bash script?

chrissierra, 2017-09-11:
Hello, 
there is a way for enable and disabled notification on motion via command line?

chrisf4lc0n, 2017-09-11:
Nope, not at the moment it is not as whatever you will put in it will get overwritten by MotionEye UI.

chrissierra, 2017-09-11:
Ok, only you can do this via interface web?

chrisf4lc0n, 2017-09-11:
Pretty much yes.

feelthepayne, 2017-09-20:
I do this with alternative config files - one with notification enabled, the other with notification turned off.  A cron job checks the arp cache on the router for the MAC address of my mobile and updates the symlink of the config file accordingly and reloads the motioneye process.

next step when I get 5 minutes is to use geofencing with ifttt and an AWS Lambda function of some sort as the ARP cache on the router can be unreliable particularly if i've turned my WiFi off for what ever reason.

chrisf4lc0n, 2017-09-20:
@feelthepayne wow! I have not thought about that. I tried it with sed command and it worked until I started the MotionEye GUI.
Really great idea mate, kudos to you!

JamBalaya56562, 2026-11-09:
Switching notifications on and off from outside the UI is requested in #1223, so closing this one as a duplicate. feelthepayne's workaround with two config files is the way to do it until then.

closed 2026-11-09 by JamBalaya56562