motioneye-project/motioneye #660: ftp server saves also on SD CARD

bigpaolo, 2017-09-11:
Hy
Thanks for ftp server on motioneye
but:
i need NOT to save on my SD CARD, but only on the FTP server.
What i gonna write on "Storage Device" and on "Root Directory" in File Storage section?
Because if i don't write anything, Motioneye request me the input of something.

Thanks

Paolo

p.s.: I have installed Motioneye on raspberry pi3 and Raspbian

chrisf4lc0n, 2017-09-12:
MotionEye synchronises the local folder with FTP so no direct save to FTP server is possible. You can however mount the remote folders through the fstab and point MotionEye to save directly to them.

bigpaolo, 2017-09-12:
Hy,
could you please explain me how to mount the external drive (mine is on 192.168.1.10/disk1/camera - port 21) in my Raspberry with Rasbpian?
Thanks
Paolo

chrisf4lc0n, 2017-09-12:
[http://www.noveldevices.co.uk/rp-cifsinfstab](http://www.noveldevices.co.uk/rp-cifsinfstab)

bigpaolo, 2017-09-15:
thanks, i'll try it

bigpaolo, 2017-09-22:
Hy,
I have mounted the external drive as you have linked.
If i go with my raspberry in the directory of the ext drive i can save a file and in windows i can found it.
In motioneye i have insert the correct link with the correct subdir and it create the correct subdir.
After that motioneye don't detect any movenent on the camera (i have selected also the "show n. of frame inthe picture" option to help me, but the number is always ZERO even if i move in front of the camera.
I try to set the n. of frame to start recording at 2 but nothing change...
What i can do?
Thanks 
Paolo

bigpaolo, 2017-09-23:
Due to this problem (i think), motioneye don't save any picture or video on my external drive. (Motioneye recognise the external drive because show the free space on it)

Thanks
Paolo

chrisf4lc0n, 2017-09-23:
Can you issue the command `sudo df -aTh` and post the results? I think you have't mounted the disks right.

bigpaolo, 2017-09-25:
Hy
here is the return of df -aTh 

File system              Tipo             Dim. Usati Dispon. Uso% Montato su
/dev/root                ext4              14G  4,7G    7,9G  38% /
devtmpfs                 devtmpfs         458M     0    458M   0% /dev
sysfs                    sysfs               0     0       0    - /sys
proc                     proc                0     0       0    - /proc
tmpfs                    tmpfs            462M   16K    462M   1% /dev/shm
devpts                   devpts              0     0       0    - /dev/pts
tmpfs                    tmpfs            462M   47M    415M  11% /run
tmpfs                    tmpfs            5,0M  4,0K    5,0M   1% /run/lock
tmpfs                    tmpfs            462M     0    462M   0% /sys/fs/cgroup
cgroup                   cgroup              0     0       0    - /sys/fs/cgroup/systemd
cgroup                   cgroup              0     0       0    - /sys/fs/cgroup/cpu,cpuacct
cgroup                   cgroup              0     0       0    - /sys/fs/cgroup/blkio
cgroup                   cgroup              0     0       0    - /sys/fs/cgroup/memory
cgroup                   cgroup              0     0       0    - /sys/fs/cgroup/devices
cgroup                   cgroup              0     0       0    - /sys/fs/cgroup/freezer
cgroup                   cgroup              0     0       0    - /sys/fs/cgroup/net_cls
systemd-1                -                   -     -       -    - /proc/sys/fs/binfmt_misc
debugfs                  debugfs             0     0       0    - /sys/kernel/debug
mqueue                   mqueue              0     0       0    - /dev/mqueue
configfs                 configfs            0     0       0    - /sys/kernel/config
fusectl                  fusectl             0     0       0    - /sys/fs/fuse/connections
/dev/mmcblk0p6           vfat              66M   22M     45M  32% /boot
//192.168.1.10/IpCamera cifs             917G   34M    917G   1% /home/pi/myNAS
tmpfs                    tmpfs             93M     0     93M   0% /run/user/1000
gvfsd-fuse               fuse.gvfsd-fuse     0     0       0    - /run/user/1000/gvfs
/dev/mmcblk0p5           ext4              30M  443K     28M   2% /media/pi/SETTINGS
binfmt_misc              binfmt_misc         0     0       0    - /proc/sys/fs/binfmt_misc

Thanks

Paolo

chrisf4lc0n, 2017-09-25:
It looks fine, can you create files/folders on the shared drive from your Pi outside MotionEye? 
Can you also post the log file, please?

bigpaolo, 2017-09-28:
Hy
Yes, i can create files and folders on the shared drive from my Pi outside Motioneye.
Is Motioneye that don't recognise the movement frames. I have choose the option "show n. of frames that change in the picture" in motion detection option and that number is always zero rven if i dance in front of the camera...!
Where i found the log file?
Paolo

JamBalaya56562, 2026-11-09:
motionEye always records locally first and then uploads, so FTP alone is not possible. To write straight to a network share, set `smb_shares true` in motioneye.conf and choose "Network Share" as the storage device, or mount the share yourself. The later detection problem got no log, and there was no news since 2017, so closing.

closed 2026-11-09 by JamBalaya56562