motioneye-project/motioneye #662: Garbage shown in RTSP

waqaslam, 2017-09-14:
Hi
I'm able to add an RTSP camera in motioneye, but the video stream shows garbage artifacts. Application is running on a Raspberry Pi3 running dietpi OS.

**motionEye Version	0.37.1
Motion Version	4.0.1
OS Version	Linux 4.9.40-v7+**

Why is this happening? and how to resolve this?
Feel free to ask for further details/logs.

Attached are the sample screenshots.
[image]
[image]
[image]

a part from motioeye.service logs:

> -- Logs begin at Thu 2017-09-14 00:14:15 CEST. --
Sep 14 14:54:44 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: cabac decode of qscale diff failed at 66 31
Sep 14 14:54:44 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 66 31, bytestream -1
Sep 14 14:54:46 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: cabac decode of qscale diff failed at 28 19
Sep 14 14:54:46 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 28 19, bytestream -1
Sep 14 14:54:50 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: left block unavailable for requested intra4x4 mode -1 at 0 20
Sep 14 14:54:50 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 0 20, bytestream 1844
Sep 14 14:54:57 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: cabac decode of qscale diff failed at 17 22
Sep 14 14:54:57 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 17 22, bytestream -1
Sep 14 14:55:07 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: cabac decode of qscale diff failed at 73 21
Sep 14 14:55:07 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 73 21, bytestream -1
Sep 14 14:55:18 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: cabac decode of qscale diff failed at 17 30
Sep 14 14:55:18 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 17 30, bytestream 0
Sep 14 14:55:28 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: cabac decode of qscale diff failed at 46 41
Sep 14 14:55:28 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 46 41, bytestream -1
Sep 14 14:55:29 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: cabac decode of qscale diff failed at 3 30
Sep 14 14:55:29 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 3 30, bytestream -1
Sep 14 14:55:36 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: left block unavailable for requested intra mode at 0 19
Sep 14 14:55:36 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 0 19, bytestream -1
Sep 14 14:55:43 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: cabac decode of qscale diff failed at 6 18
Sep 14 14:55:43 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 6 18, bytestream -1
Sep 14 14:55:53 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: cabac decode of qscale diff failed at 60 29
Sep 14 14:55:53 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 60 29, bytestream -1
Sep 14 14:56:05 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: cabac decode of qscale diff failed at 41 33
Sep 14 14:56:05 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 41 33, bytestream -1
Sep 14 14:56:11 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: cabac decode of qscale diff failed at 39 30
Sep 14 14:56:11 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 39 30, bytestream -1
Sep 14 14:56:11 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: cabac decode of qscale diff failed at 40 21
Sep 14 14:56:11 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 40 21, bytestream -1
Sep 14 14:56:20 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: cabac decode of qscale diff failed at 6 5
Sep 14 14:56:20 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 6 5, bytestream -1
Sep 14 14:56:25 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: cabac decode of qscale diff failed at 76 15
Sep 14 14:56:25 pi-server motion[15058]: [0:av1] [ERR] [ENC] ffmpeg_avcodec_log: error while decoding MB 76 15, bytestream -1

ccrisan, 2017-09-14:
Please report this to [motion](https://github.com/Motion-Project/motion).

closed 2017-09-14 by ccrisan

waqaslam, 2017-09-14:
The issue is also reported on **motion**.
https://github.com/Motion-Project/motion/issues/503

waqaslam, 2017-09-14:
The problem is fixed as suggested my motion support guys:

My input source has a native resolution **1280x720** but in motioneye's settings it was set to **640x480**.
Setting the correct resolution (i.e. 720p) fixed the problem.

ccrisan, 2017-09-14:
Thanks for reporting back!