motioneye-project/motioneye #663: NOIR and wifi fail after web update to MEyeOS 20170710

dbgarlisch, 2017-09-17:
I was running MotionEyeOS v 20170212 without issue.
NOIR Camera worked. Wireless connection worked. All was working.

I used the web config to check for Updates. It found "New version available: 20170710. Update?"
I selected YES. It churned for a few minutes and said the update completed.

After rebooting, the wireless connection fails and it cannot detect the NOIR camera anymore. Nothing I have tried has worked.

An interesting note, afyer the update, the MotionEye web interface still reports the current version as:
   motionEye Version	20170212
   Motion Version	4.0.1+git37b3595

Any suggestions? Or am I doomed to just do a fresh install of the 20170710 update?

David "If it ain't broke don't fix it" G.

ccrisan, 2017-09-18:
@dbgarlisch unfortunately updating from a relatively old version could in some cases break the system. You could try repeating the update but I believe you'll have to reinstall everything from scratch. The update mechanism has been improved with 20170710 so it's unlikely problems like this would happen again. I'm sorry for the inconvenience.

closed 2017-09-18 by ccrisan

dbgarlisch, 2017-09-18:
Thanks for the quick reply.
Just so you know...
I did a fresh install of 20170710 and all is happy again.