motioneye-project/motioneye #664: MotionEye does not show camera images when I access from Internet!

farahpoor, 2017-09-18:
I install MotionEye on RPI and it works well,first of all thanks for your great job Calin,
but I forwarded the proper ports in my router in order to access to it from Internet, and I can access to the motioneye and its environments but it does not show camera image! how ever when I access it from local network every things is fine.
what is the problem?
the image is below:
[https://preview.ibb.co/gMyaeQ/123.png](url)

ccrisan, 2017-09-18:
What kind of camera is this?

farahpoor, 2017-09-18:
RPI CSI Camera(Onboard)
[Camera](https://statics3.seeedstudio.com/product/Pi%20Camera_04.jpg)

ccrisan, 2017-09-18:
Have you enabled Fast Network Camera?

farahpoor, 2017-09-18:
it does not show anything either enable and disable  mode.
however in local network both modes work well.

ccrisan, 2017-09-18:
I have no idea then. Sorry.

farahpoor, 2017-09-18:
for a time you can access it here: http://2.182.7.187:8079/

farahpoor, 2017-09-18:
user pass
admin,admin

Firesphere, 2017-09-18:
Do you happen to have HTTPS enabled on your external port?

MrSimonC, 2018-01-22:
For those reading this, I had the exact same issue, but have it working. 
If doing port forwarding through your router, then you can access the feed directly by using port 8081. e.g. port forward a port into yourMotionEyeDevice:8081.

If using SSH SOCKS / port forwarding via ssh, you can use this SSH command on the device (external to your home network):
`ssh -L 127.0.0.1:8080:myMotionEyeDevice:8081 username@myDynamicHomeRouterAddr.ess`
then open `127.0.0.1:8080` on your browser (change 8080 to a port of your preference).

electron1979, 2018-12-10:
I have port-forwarded my motionOS page, so I can access it remotely:
My Android cams (using "IP Webcam" app) show-up internally, but not from outside my network/firewall (running openWRT on my router). Same issue as OP!

Tinkernut did a guide [here](https://youtu.be/L9_imMScblE)!

JamBalaya56562, 2026-11-09:
As MrSimonC found, when the browser shows the camera through Motion's streaming port (8081 by default), that port has to be forwarded as well, or reached through a tunnel. Closing, as this answers it.

closed 2026-11-09 by JamBalaya56562