motioneye-project/motioneye #665: Cant add RTSP camera motioneye/Ubuntu16.04

gregewing, 2017-09-18:
Hi.
I've installed a couple times now and still cannot get motioneye (or Motion for that matter) to accept a feed from an rtsp streaming camera on the local LAN.   Interestingly, motioneyeOS has no such problems.

Please can you suggest diagnostics or fixes?

error log messages are as follows:
cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf

chrisf4lc0n, 2017-09-18:
I cannot see any error messages...

gregewing, 2017-09-18:
rookie copy-n-paste error in haste.  apologies.

messages as follows:
/var/log/motion.log:
[21852544] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[21852544] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[21852544] [NTC] [ALL] motion_startup: Motion 3.2.12+git20140228 Started with SDL support
[21852544] [NTC] [ALL] motion_startup: Logging to syslog
[21852544] [NTC] [ALL] motion_startup: Using default log type (ALL)
[21852544] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.1.130/Streaming/Channels/1)
[1] [ALR] [NET] rtsp_connect: unable to open input(rtsp://192.168.1.130:554/Streaming/Channels/1): -825242872 - Server returned 401 Unauthorized (authorization failed)

snip from /var/log/motion/motion.log:
[33456512] [NTC] [ALL] [Sep 18 16:28:14] motion_startup: Using log type (ALL) log level (NTC)
[33456512] [NTC] [ALL] [Sep 18 16:28:14] become_daemon: Motion going to daemon mode
[33456512] [NTC] [ALL] [Sep 18 16:28:14] become_daemon: Created process id file /var/run/motion/motion.pid. Process ID is 28592
[33456512] [NTC] [ALL] [Sep 18 16:28:14] motion_startup: Motion running as daemon process
[33456512] [NTC] [ENC] [Sep 18 16:28:14] ffmpeg_init: ffmpeg LIBAVCODEC_BUILD 3685476 LIBAVFORMAT_BUILD 3680357
[0] [NTC] [ALL] [Sep 18 16:28:14] main: Thread 1 is from /etc/motion/motion.conf
[0] [NTC] [ALL] [Sep 18 16:28:14] main: Thread 1 is device: rtsp://admin:th1ssux@192.168.1.130:554/video.h264 input -1
[0] [NTC] [ALL] [Sep 18 16:28:14] main: Stream port 8081
[0] [NTC] [ALL] [Sep 18 16:28:14] main: Waiting for threads to finish, pid: 28592
[0] [NTC] [STR] [Sep 18 16:28:14] httpd_run: motion-httpd testing : IPV4 addr: 0.0.0.0 port: 8085
[0] [NTC] [STR] [Sep 18 16:28:14] httpd_run: motion-httpd Bound : IPV4 addr: 0.0.0.0 port: 8085
[0] [NTC] [STR] [Sep 18 16:28:14] httpd_run: motion-httpd/3.2.12+git20140228 running, accepting connections
[0] [NTC] [STR] [Sep 18 16:28:14] httpd_run: motion-httpd: waiting for data on 0.0.0.0 port TCP 8085
[1] [NTC] [ALL] [Sep 18 16:28:14] motion_init: Thread 1 started , motion detection Enabled
[1] [ALR] [NET] [Sep 18 16:28:14] netcam_start: Network Camera thread starting... for url (rtsp://admin:th1ssux@192.168.1.130:554/video.h264)
[1] [ALR] [NET] [Sep 18 16:28:14] rtsp_connect: unable to open input(rtsp://192.168.1.130:554/video.h264): -825242872 - Server returned 401 Unauthorized (authorization failed)
[17867136] [NTC] [ALL] [Sep 18 16:28:36] motion_startup: Using log type (ALL) log level (NTC)
[17867136] [NTC] [ALL] [Sep 18 16:28:36] become_daemon: Motion going to daemon mode
[17867136] [NTC] [ALL] [Sep 18 16:28:36] become_daemon: Created process id file /var/run/motion/motion.pid. Process ID is 28692
[17867136] [NTC] [ALL] [Sep 18 16:28:36] motion_startup: Motion running as daemon process
[17867136] [NTC] [ENC] [Sep 18 16:28:36] ffmpeg_init: ffmpeg LIBAVCODEC_BUILD 3685476 LIBAVFORMAT_BUILD 3680357
[0] [NTC] [ALL] [Sep 18 16:28:36] main: Thread 1 is from /etc/motion/motion.conf
[0] [NTC] [ALL] [Sep 18 16:28:36] main: Thread 1 is device: rtsp://admin:th1ssux@192.168.1.130:554/video.h264 input -1
[0] [NTC] [ALL] [Sep 18 16:28:36] main: Stream port 8081
[0] [NTC] [ALL] [Sep 18 16:28:36] main: Waiting for threads to finish, pid: 28692
[1] [NTC] [ALL] [Sep 18 16:28:36] motion_init: Thread 1 started , motion detection Enabled
[1] [ALR] [NET] [Sep 18 16:28:36] netcam_start: Network Camera thread starting... for url (rtsp://admin:th1ssux@192.168.1.130:554/video.h264)
[0] [NTC] [STR] [Sep 18 16:28:36] httpd_run: motion-httpd testing : IPV4 addr: 0.0.0.0 port: 8085
[0] [NTC] [STR] [Sep 18 16:28:36] httpd_run: motion-httpd Bound : IPV4 addr: 0.0.0.0 port: 8085
[0] [NTC] [STR] [Sep 18 16:28:36] httpd_run: motion-httpd/3.2.12+git20140228 running, accepting connections
[0] [NTC] [STR] [Sep 18 16:28:36] httpd_run: motion-httpd: waiting for data on 0.0.0.0 port TCP 8085
[1] [ALR] [NET] [Sep 18 16:28:36] rtsp_connect: unable to open input(rtsp://192.168.1.130:554/video.h264): -825242872 - Server returned 401 Unauthorized (authorization failed)

gregewing, 2017-09-18:
It looks like authorisation failed, but the exact same URL works in vlc
The ID and PW are correct, but maybe not being passed right?  How should they be formatted in motion.conf as the suggested and normal formats don't work.

Should I be using the precompiled ffmpeg binaries or those from apt?

gregewing, 2017-09-18:
Ok, so I installed the version 4.0.1 of motion daemon from GitHub and now everything is working fine.  Please close this ticket.

chrisf4lc0n, 2017-09-19:
Since you have started it, you can close the ticket yourself.

gregewing, 2017-09-19:
Issue Resolved - use the pre-build motion binaries, not the ones from Canonical.

closed 2017-09-19 by gregewing