motioneye-project/motioneye #666: USB Webcam not detected

chrisberesford, 2017-09-19:
I have a Microsoft LifeCam HD-3000 which is recognised by ubuntu and no errors are listed on dmesg.
When I go to add camera in the drop down lookup in motioneye, it says that no local cameras are detected. I can see from past issues logged here that users have had success with this model of camera, and it is working on my mac. Is there anything I can check to investigate why it's not displayed in motioneye?
Thanks

chrisf4lc0n, 2017-09-20:
A full log from the moment you are trying to add the camera please.

chrisberesford, 2017-09-20:
Here is the output from dmesg:
[  198.061338] usb 1-2.1: new high-speed USB device number 5 using uhci_hcd
[  198.442421] usb 1-2.1: New USB device found, idVendor=045e, idProduct=0779
[  198.442424] usb 1-2.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[  198.442426] usb 1-2.1: Product: Microsoft® LifeCam HD-3000
[  198.442428] usb 1-2.1: Manufacturer: Microsoft
[  198.448432] uvcvideo: Found UVC 1.00 device Microsoft® LifeCam HD-3000 (045e:0779)
[  198.459263] input: Microsoft® LifeCam HD-3000 as /devices/pci0000:00/0000:00:11.0/0000:02:03.0/usb1/1-2/1-2.1/1-2.1:1.0/input/input7

Nothing is written to the motioneye log. The USB camera just never appears in the lookup for local cameras.

chrisf4lc0n, 2017-09-20:
Issue motion -h command please.

chrisberesford, 2017-09-20:
motion -h shows the help:
`motioneye@ubuntu:~/motioneye$ motion -h
motion Version 3.2.12+git20140228, Copyright 2000-2005 Jeroen Vreeken/Folkert van Heusden/Kenneth Lavrsen

usage:	motion [options]

Possible options:

-n			Run in non-daemon mode.
-s			Run in setup mode.
-c config		Full path and filename of config file.
-d level		Log level (1-9) (EMR, ALR, CRT, ERR, WRN, NTC, ERR, DBG, ALL). default: 6 / NTC.
-k type			Type of log (COR, STR, ENC, NET, DBL, EVT, TRK, VID, ALL). default: ALL.
-p process_id_file	Full path and filename of process id file (pid file).
-l log file 		Full path and filename of log file.
-m			Disable motion detection at startup.
-h			Show this screen.

Motion is configured using a config file only. If none is supplied,
it will read motion.conf from current directory, ~/.motion or /etc/motion.

`

chrisf4lc0n, 2017-09-20:
Install motion 4.x and try it again.

chrisberesford, 2017-09-20:
`motioneye@ubuntu:~$ motion -h
motion Version 4.0.1, Copyright 2000-2016 Jeroen Vreeken/Folkert van Heusden/Kenneth Lavrsen/Motion-Project maintainers

usage:	motion [options]

Possible options:

-b			Run in background (daemon) mode.
-n			Run in non-daemon mode.
-s			Run in setup mode.
-c config		Full path and filename of config file.
-d level		Log level (1-9) (EMG, ALR, CRT, ERR, WRN, NTC, INF, DBG, ALL). default: 6 / NTC.
-k type			Type of log (COR, STR, ENC, NET, DBL, EVT, TRK, VID, ALL). default: ALL.
-p process_id_file	Full path and filename of process id file (pid file).
-l log file 		Full path and filename of log file.
-m			Disable motion detection at startup.
-h			Show this screen.

Motion is configured using a config file only. If none is supplied,
it will read motion.conf from current directory, ~/.motion or /etc/motion.
`
USB camera still does not appear in the lookup. It is listed in lsusb. Could this be a permissions problem? I'm running motioneye as the user 'motioneye' from that user's home directory.

chrisf4lc0n, 2017-09-20:
Might be a silly question but have you restarted motioneye process after installing the new motion?

chrisberesford, 2017-09-20:
No question is silly.......
I have restarted the whole Ubuntu VM.
I have tested the camera on the host using cheese and it works fine.

chrisf4lc0n, 2017-09-20:
VM... I am not sure how motion interacts with VM USB devices... I ran WiFi cameras on the VM but never tried the USB ones...

ccrisan, 2017-09-20:
@chrisberesford:

1. Do you have any other cameras added to your motionEye system?
2. Have you tried with any other camera (model) just to see if it works? If not, could you do that?
3. What does `ls -l /dev/video*` say?
4. What does `v4l2-ctl --list-devices` say? Make sure to run it with the user with which motionEye runs.
5. You may want to set the log level to DEBUG in your `motioneye.conf` and try again. The log file generated could contain useful details.

chrisberesford, 2017-09-25:
After a bout of food poisoning, I've been able to have a look at this a bit more. A spare Pi3 and a fresh image, with the same camera works fine, so there is some issue with ubuntu, or the motioneye version for ubuntu.

I will be happy to plug the camera back into the ubuntu VM and try again, but it will not be for a couple of days. I need the camera working right now to keep an eye on my construction work.

chrisf4lc0n, 2017-09-26:
I don't think that there is a problem with Ubuntu or MotionEye on Ubuntu. The problem lies with the way VM interacts with USB devices, try to play around with it. I have been running Ubuntu with MotioEye for years with various cameras and never encountered a problem with their detection yet.

JamBalaya56562, 2026-11-09:
The same camera worked on a Pi, so this points at how the VM passes the USB camera through rather than at motionEye. If a camera that `v4l2-ctl --list-devices` shows is still missing in a current version, please open a new issue. Closing.

closed 2026-11-09 by JamBalaya56562