motioneye-project/motioneye #667: raspberry pi 3 camera not detected by motioneye

cahmedc, 2017-09-22:
[image]

I have tried many things 
appreciate your support ..

Regards

chrisf4lc0n, 2017-09-22:
Issue command `motion -h` please.

wb666greene, 2017-09-22:
Raspbian Stretch or Jessie?
V1 or V2 pi camera?

I have Motioneye running on Stretch with a PiZero-W and a V2 camera and don't recall doing anything to get the pi camera to appear in the add camera dialog.
  
Unfortunately the PiZero-W (and Pi3) WiFi chip has firmware issues that make the Pi-Zero unusable with Motioneye as the WiFi drops out under "heavy load" or "weak signal" conditions and needs a reboot to recover.  The Pi3 can run with wired networking, but is reported to have the same WiFi issues, this issue is over a year old and the fix is utterly dependent on the chip vendor to do a firmware update  :(

I also installed Motioneye on a Pi2 running Stretch with a V1 camera and the pi camera did not show up until I did:
sudo nano /etc/modules
// add:
bcm2835-v4l2

and rebooted.

The Pi2 with a WiPi WiFi USB dongle has been running Motioneye for almost two days now, with 2592x1944 still images without issues.  After a few more days of testing, I'll disable the on-board WiFi of the PiZero-W and use the WiPi dongle with an OTG cable and seem if that is a viable work-around.

However, my Pi2  v4l2-ctl --list-devices result looks exactly like yours does (minus the USB camera as I've only the Pi Camera), unfortunately I never ran it before adding the line to /etc/modules.

cahmedc, 2017-09-23:
@wb666greene tested on v1 camera, pi 2, Jessie  before and it was working 100% 
but v2 camera pi3 Jessie,  it's not working, no recognition.

wb666greene, 2017-09-23:
This may be too basic, but I've had troubles with the Pi Camera cable connections.  Does raspistill -k capture images on your Pi3?
Your problems are magnified if running headless, scrounge up an HDMI monitor, mouse, & keyboard, as otherwise quick diagnostics like this are more difficult.

If it does show the camera preview, have you added the v4l2 driver line in /etc/modules?  I think Motioneye tries to do it, but seems it can fail.

Have you tried booting without the USB web cam plugged in?  Does your Motioneye setup offer the USB camera to you?  Does it work if you select it?

I'm no Motioneye expert, but once I got it installed on Stretch it has worked really well.

cahmedc, 2017-09-23:
yea its capturing images and videos. 
I'm not using anything else on the board ( no mouse or keyboard or HDMI ), the only ssh.
Yes, I have added v412 to /etc/modules .
Yes I have tried it without USB webcam.
Motioneye doesnt recognise USB cam too.

I had good time with motioneye on pi2 . 

Really appreciate your cooperation ..
Thanks @wb666greene

cahmedc, 2017-09-24:
[image]

[image]

Is this related to the issue ? 
just trying new installation of Stretch on pi3

cahmedc, 2017-09-25:
after following  @wb666greene on this link https://github.com/ccrisan/motioneye/issues/635
issue solved.
Thanks a lot.

closed 2017-09-26 by ccrisan