motioneye-project/motioneye #669: How to stream with motioneye trough a GSM proxy/ firewall?

Alpengroove, 2017-09-24:
Hi!

I use motioneyeos as a surveillance camera on my boat. I use a GSM network, so I can't  forward ports.
I can connect the pi via a cloud based RealVNC client, that tunnels through the GSM net.

Is there a possibility to stream forward with motioneye to a service (free would be great), and to get the stream from this service?

best regards,
Martin

alxlo, 2017-10-09:
Either you can set up a reverse SSH tunnel or a VPN or - if you don't want to set up and maintain an additional system for this - you might set this up as a tor hidden service (aka poor mans vpn). It should work like https://www.parckwart.de/?page=computer_stuff&article=vnc_remote_support_via_tor_onion_services but instead making the vnc port available, make your streaming urls available as a hidden service.

Alpengroove, 2017-10-10:
Thank you for your advice.
Meanwhile I managed to solve the remote access with the ngrok cloud based tunneling service.
It took. 3 min to install an wotks perfect.

Martin

tehKaiN, 2018-01-01:
I've used Logmein Hamachi with my OrangePi and Odroid C2 with success. It's free, works on x86 and ARM archs too.

Dararii, 2019-12-06:
@Alpengroove can you give some tutorial on how to set up ngrok in motioneyeos? Thanks in advance.

JamBalaya56562, 2026-11-07:
Closing as solved, as you got remote access working with the ngrok tunnelling service. The comments above also list other ways, such as a reverse SSH tunnel or a VPN.

closed 2026-11-07 by JamBalaya56562