motioneye-project/motioneye #671: saving files to usb-stick

Tinker-Bel, 2017-09-25:
Hey,
I have a setup with a 'master' ME-os and a 'slave' ME camera.
I save the motion triggered movies on a usb-stick connected to the 'slave' ME camera.
When i disconnect the usb-stick and put ik back in again, the newly created motion triggered movies aren't saved on the usb-stick.
I've tried resetting the storage device and root directory, rebooting,.... but with nog luck.

any idea what I'm doing wrong?

Kind regards,
Christophe

chrisf4lc0n, 2017-09-26:
Most of the Linuxes do not auto mount the USB drives on plugging in. You would need to configure that via usbmount or autofs, plus I believe you would need to restart the motioneye process once done that.

JamBalaya56562, 2026-11-09:
As answered, most Linux systems do not mount a USB stick again by themselves after it is plugged back in; usbmount or autofs (or an fstab entry with `nofail`) does that, then restart motionEye. Closing.

closed 2026-11-09 by JamBalaya56562