motioneye-project/motioneye #673: Erorr on banana pi-m64

dntwrd, 2017-09-27:
Why does not the "motioneye" see the webcam?
[image]
[image]

ccrisan, 2017-09-27:
what does v4l2-ctl --list-devices say?

dntwrd, 2017-09-28:
[image]

dntwrd, 2017-09-29:
who can help me?

ccrisan, 2017-09-30:
@dntwrd I have no idea what "the XX" is, but it's good that you can now add your camera. If there's no video, please **attach** here (via github, not email) your motion.log file.

dntwrd, 2017-09-30:
where is the motion.log?

ccrisan, 2017-10-01:
Normally at `/var/log/motion.log`.

dntwrd, 2017-10-01:
`/usr/local/bin/motion: error while loading shared libraries: libjpeg.so.62: cannot open shared object file: No such file or directory
`

dntwrd, 2017-10-01:
If I write "sudo apt-get install jpeg62", I get this error
[image]

ccrisan, 2017-10-02:
Well this is not a motionEye issue. Make sure your motion installation works.

closed 2017-10-02 by ccrisan