motioneye-project/motioneye #674: Better mail sending (use sendmail, send camera lost warnings, be able to attach movie)

Harvie, 2017-09-29:
Hello,
please add way to globaly configure mail sending. I've been using following configuration of motion (without motioneye) and i can't get similar setup with motioneye:

```
on_picture_save echo motion | mutt -e 'set copy=no' -a %f -s "motion detected (camera%t picture)" -- foo@bar.baz, bah@ban.lol
on_movie_end echo motion | mutt -e 'set copy=no' -a %f -s "motion detected (camera%t movie)" -- foo@bar.baz, bah@ban.lol
on_camera_lost echo camera%t lost | mail -s "camera%t lost" -- foo@bar.baz, bah@ban.lol
```

1.) it used local binaries like mutt, sendmail and mail to send the email, therefore there was no need to configure anything. Please add option to "use local MTA", it simplifies configuration on servers that already run full mailserver. (right now i had to create postfix user and use localhost with login in motioneye)

2.) It was able to send me email if something went wrong with camera (camera lost)

3.) It sent me SWF every time there was movement and it was easy for me to play it directly in roundcube webmail. Also there was no need to take still images in adition to movies just to be sent in email. Movies are simply better. However sometimes they can be too big for email. But i can live with that as long as i manage my own mailserver and therefore i can handle that. This is not exactly good for all cases, but i'd like to have this as advanced option. idealy if it was possible to cut just few seconds of movie, so it was not bigger than 8MB or any configured size.

ccrisan, 2017-09-30:
1. You can simply use the File Storage > Run A Command option for both `on_picture_save` and `on_movie_end` events to run your own script that will do the notifications.
2. The email sending mechanism available out-of-the-box in motionEye is intended for users that don't know or don't care about writing scripts, installing programs and designing shell commands. For them the email setup as it is now just works and is good enough. For guys like you, there's always the Run A Command option (available under Motion Notifications as well, by the way).
3. There are plans to make the movie playable directly from the browser. There's a PR about this (#218). However it's going to work from the UI, not via email or a direct link or something like that.

closed 2017-09-30 by ccrisan

Harvie, 2017-09-30:
1.) I don't want to write scripts, i would be more than happy if that just worked out of box with minimal config. I guess motioneye should try to use "sendmail" or "mail" binary when no remote smtp server is setup. That's what most of the daemons use and it's great for everyone, since it requires ZERO CONFIGURATION.

2.) But what's wrong about sending email when camera gets disconnected? That might also be usefull for regular users.

3.) Well i guess once player will be implemented it should be very easy to send URL of that player with any video and put it in the email.

ccrisan, 2017-10-01:
1. I have very few experience with `sendmail` or `mail`. Also I believe very few motionEye users have any of those configured on their system. I still believe it is not worth adding them to the UI as alternatives to the SMTP method.
2. Nothing is wrong about that, it's just that the event is not covered by the UI. You'll have to use the Extra Motion Options field to configure that. I agree that the UI could support this event in a way or another.

Harvie, 2017-10-01:
> I have very few experience with sendmail or mail

I've never really programmed in python, but i've found two code examples on https://stackoverflow.com/questions/73781/sending-mail-via-sendmail-from-python

```
from email.mime.text import MIMEText
from subprocess import Popen, PIPE

msg = MIMEText("Here is the body of my message")
msg["From"] = "me@example.com"
msg["To"] = "you@example.com"
msg["Subject"] = "This is the subject."
p = Popen(["/usr/sbin/sendmail", "-t", "-oi"], stdin=PIPE, universal_newlines=True)
p.communicate(msg.as_string())
```
OR

```
def sendMail():
    sendmail_location = "/usr/sbin/sendmail" # sendmail location
    p = os.popen("%s -t" % sendmail_location, "w")
    p.write("From: %s\n" % "from@somewhere.com")
    p.write("To: %s\n" % "to@somewhereelse.com")
    p.write("Subject: thesubject\n")
    p.write("\n") # blank line separating headers from body
    p.write("body of the mail")
    status = p.close()
    if status != 0:
           print "Sendmail exit status", status
```

Good think about this approach is that on typical Linux server user does not need to know sh*t about mailservers or logins. All user needs to set is destination mail address and it just works. However on laptops and other user stations it still might need to use external SMTP server with login, but this is good default setting that will just work without further configuration in lots of cases...