motioneye-project/motioneye #675: Enable more stuff to be configured globally

Harvie, 2017-09-29:
If you have more cams it's PITA to configure them individualy. When you are using motion without motioneye, you can put everything into motion.conf and it will be inherited in all threads as defaults unless overrided specificaly. So you just setup name of each camera and how to access it. Everything else can be inherited from global defaults, including:

* motion detection
* recording and snapshotting details (filenames, intervals, picture quality, filters, etc...)
* streaming
* event hooks (sending emails, files,...)

ccrisan, 2017-09-30:
What you're asking here makes sense but can't be easily implemented given the way motionEye handles the configuration. One idea would be to choose a camera and an option and to have the possibility to apply it across all (or some of the) cameras.

Harvie, 2017-09-30:
i understood that it's stored directly in motion.conf and thread-*.conf, so it should be quite easy to add all of this stuff to motion.conf the same way you store it in thread-1.conf. Then you can easily delete the setting from thread-1.conf if user will want to inherit global settings for that option. You will know it's not overriden simply by the fact it will be missing in thread-1.conf which you can indicate in UI like "default" or by graying out entire section. Once user will set something different from "default" and therefore override globals it should be ungrayed in UI and re-added to thread-1.conf.