motioneye-project/motioneye #677: Do not set text_double on for cams 800x600 and smaller

Harvie, 2017-09-30:
Do not set `text_double on` for cams 800x600 and smaller. It does not make any sense.

ccrisan, 2017-09-30:
Currently all cameras with a width above 320 have `text_double on`. I personally found this the best compromise, given the only two text sizes we have. Do you recommend that we increase the threshold from 320 to 800?

Harvie, 2017-09-30:
on 640x480 it's too big. Don't know about 800x600, it's probably in the middle.

ccrisan, 2017-10-01:
Implemented via eb92d5636a741ea5534e17eebae44a133a849b64.

closed 2017-10-01 by ccrisan

arkenoa, 2018-06-13:
I'm watching the MotionEye interface primarily from from my phone, and on my 640x480 cameras the smaller text is just unreadable. Now every time I made changes from the UI, I must manually edit thread1 and 2 conf files to adjust text_double on manually.
May be the best decision is to just add an option to set the text size in "text overlay" section in the UI.

Harvie, 2018-06-14:
@arkenoa maybe just enable double text globally?

anyway best solution would obviously be to patch motion to be able to set text size as percentage of image size instead of constant pixel height. Would you mind opening such ticket in motion repository?