motioneye-project/motioneye #679: Fullscreen not working when fit frames = on and layour rows > 1

Harvie, 2017-09-30:
i set fit frames verticaly = on and layout rows = 2 (in chrome). Then i click on fullscreen preview of cam and it goes fullscreen, but camera does not expand over screen:
[image]

it works with layout rows = 1.

ccrisan, 2017-10-01:
This is a known bug. It has been briefly mentioned in https://github.com/ccrisan/motioneyeos/issues/1134.

JamBalaya56562, 2026-11-13:
This works on the current dev: with *Fit frames vertically* on and 2 layout rows, the full screen button makes the camera fill the window (checked with two cameras in a 1024x768 browser window).

closed 2026-11-13 by JamBalaya56562