motioneye-project/motioneye #680: install motioneye on Raspbian stretch give some errors

bigpaolo, 2017-10-01:
Hy 
i have installed a new Raspberry with Raspbian stretch,
whe i use the instruction to install motioneye the sysem give  lot of error:

1 install motion:

pi@raspberrypi:~ $ sudo dpkg -i pi_stretch_motion_4.0.1-1_armhf.deb
(Lettura del database... 123169 file e directory attualmente installati.)
Preparativi per estrarre pi_stretch_motion_4.0.1-1_armhf.deb...
Estrazione di motion (4.0.1-1) su (4.0.1-1)...
dpkg: problemi con le dipendenze impediscono la configurazione di motion:
 motion dipende da libmariadbclient18 (>= 5.5.36); comunque:
  Il pacchetto libmariadbclient18 non è installato.
 motion dipende da libpq5; comunque:
  Il pacchetto libpq5 non è installato.

dpkg: errore nell'elaborare il pacchetto motion (--install):
 problemi con le dipendenze - lasciato non configurato
Elaborazione dei trigger per systemd (232-25+deb9u1)...
Elaborazione dei trigger per man-db (2.7.6.1-2)...
Si sono verificati degli errori nell'elaborazione:
 motion

2 install motioneye:

sudo pip install motioneye
Collecting motioneye
  Downloading motioneye-0.37.1.tar.gz (754kB)
    100% |████████████████████████████████| 757kB 290kB/s
Requirement already satisfied: jinja2 in /usr/lib/python2.7/dist-packages (from motioneye)
Requirement already satisfied: pillow in /usr/lib/python2.7/dist-packages (from motioneye)
Collecting pycurl (from motioneye)
  Downloading pycurl-7.43.0.tar.gz (182kB)
    100% |████████████████████████████████| 184kB 1.1MB/s
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-H_0bGY/pycurl/setup.py", line 823, in <module>
        ext = get_extension(sys.argv, split_extension_source=split_extension_source)
      File "/tmp/pip-build-H_0bGY/pycurl/setup.py", line 497, in get_extension
        ext_config = ExtensionConfiguration(argv)
      File "/tmp/pip-build-H_0bGY/pycurl/setup.py", line 71, in __init__
        self.configure()
      File "/tmp/pip-build-H_0bGY/pycurl/setup.py", line 107, in configure_unix
        raise ConfigurationError(msg)
    __main__.ConfigurationError: Could not run curl-config: [Errno 2] No such file or directory

    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-H_0bGY/pycu

3 and then:

pi@raspberrypi:~ $ sudo cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/mo                                                      tioneye/motioneye.conf
cp: impossibile eseguire stat di '/usr/local/share/motioneye/extra/motioneye.conf.sample'                                                      : File o directory non esistente

4 and then:

pi@raspberrypi:~ $ cp /usr/local/share/motioneye/extra/motioneye.systemd-unit-local /etc/systemd/system/motioneye.service                      cp: impossibile eseguire stat di '/usr/local/share/motioneye/extra/motioneye.systemd-unit-local': File o directory non es                      istente

can someone help me please?

Paolo

Sailsoft2015, 2017-10-08:
Same here.  I love using motioneye for quiet some time now.
After a power failure my raspberry flash got corrupted, so I had to start from scratch.

Cost me all day to get things working again, since the installation procedure is not working on Raspbian stretch. Lots and lots of dependency errors. Especially step 2 of the ◦Install On Raspbian instruction.
Several packages do not exist anymore, or depend on other not existing packages.

This could scare of new users...

wb666greene, 2017-10-12:
Answered here last month:
https://groups.google.com/forum/#!topic/motioneye/wxdFOn2a28M

Also answered here #635 

Ask the author of the Raspbian Instructions to update his instructions for stretch.

gstevenson, 2017-11-21:
Thanks @wb666greene, just came across your post and helped me on a reinstallation after installing a new memory card in my pi.

electron1979, 2018-12-10:
Close???

closed 2018-12-17 by ccrisan

AdityaCH1997, 2019-01-26:
I cannot create the configuration directory.
//cp /usr/local/share/motioneye/extra/motioneye.conf.sample /etc/motioneye/motioneye.conf
//cp:  cannot stat  '/usr/local/share/motioneye/extra/motioneye.conf.sample':  No such file or directory.
However, installation took place perfectly with pip automaticlly pulling the Python dependancies.
Please help.

ccrisan, 2019-01-26:
You might have installed motionEye on python3, or you might need to remove the `local` part of the path.