motioneye-project/motioneye #681: CentOS installation issues

rafi1975, 2017-10-01:
Hi,

I've been trying to install motioneye on Centos 7. I followed all instructions and when trying to start it I'm getting the following errror:

  File "/usr/bin/meyectl", line 9, in <module>
    load_entry_point('motioneye==0.37.1', 'console_scripts', 'meyectl')()
  File "/usr/lib/python2.7/site-packages/motioneye/meyectl.py", line 254, in main
    import server
  File "/usr/lib/python2.7/site-packages/motioneye/server.py", line 28, in <module>
    from tornado.ioloop import IOLoop
  File "/usr/lib64/python2.7/site-packages/tornado/ioloop.py", line 47, in <module>
    from tornado.concurrent import TracebackFuture, is_future
  File "/usr/lib64/python2.7/site-packages/tornado/concurrent.py", line 32, in <module>
    from tornado.log import app_log
  File "/usr/lib64/python2.7/site-packages/tornado/log.py", line 37, in <module>
    from tornado.escape import _unicode
  File "/usr/lib64/python2.7/site-packages/tornado/escape.py", line 27, in <module>
    from tornado.util import unicode_type, basestring_type, u
ImportError: cannot import name u

Any help?

Thanks,

Rafal

rafi1975, 2017-10-02:
Any one to help on this? I just did the install from scratch and it's still the same error. When installing it says the system meets all requirements but the service fails to start with the error above. 

Thanks

donvipre, 2017-10-29:
i installed it today on centos7.
How can I help?
which installation method did you used?
epel + rpmfusion repository (second one for ffmpeg)
yum install python-pip
pip install motioneye

marksie1988, 2019-04-25:
easy fix for this, uninstall tornado and then pip install tornado :)

closed 2022-03-19 by MichaIng