motioneye-project/motioneye #682: Directory for photos not changing

karl69942, 2017-10-02:
Hey, so I changed the target directory for the photos but its not actually changing.. I don't know how to produce a log or whatever to give to you or I would do that right off the bat, sorry. I have been messing around with it all day & it doesn't wanna change.. I had it working before just fine! I don't understand why its not working anymore...

# default output path for media files (must be writable by motionEye)
media_path /home/kzp/Mounted/PiStorage

ccrisan, 2017-10-02:
I don't unserstand what your problem is. You can change the **File Storage > Root Directory** from the UI. The `media_path` option only controls the **default** path.

closed 2017-10-02 by ccrisan

karl69942, 2017-10-02:
:S where on the UI is this?

karl69942, 2017-10-02:
I don't see this option anywhere... I can't even make it automatically record movement either.. Is there a menu for the UI somewhere that I'm not seeing? :/

ccrisan, 2017-10-02:
You probably are logged in with normal user. Make sure you're logged in with admin.

karl69942, 2017-10-02:
I am logged in with admin though!

karl69942, 2017-10-02:
https://imgur.com/a/8RZUO

karl69942, 2017-10-02:
ohh... I hit advanced and everything changed... I didn't know that!! Sorry! 
This is a really cool program you made, sorry to have bothered you, I seriously didn't know all that stuff changed when I click advanced mode, I only saw a few things change before & didn't realize it gave me all these other options.. really sorry! Thanks for your help!