motioneye-project/motioneye #685: Can't access picture gallery after deleting some pictures via terminal

andrin55, 2017-10-07:
Hello

After I deleted some pictures from the filesystem, I'm no longer able to open the gallery from the webinterface:
[image]

Even if I change the directory from the web interface to a completely new one, it doesn't work.
Here is the log:
meyectl[283]:     INFO: hello! this is motionEye server 0.37.1
motion[297]: [4409184:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
motion[297]: [4409184:motion] [WRN] [ALL] thread config option deprecated use camera
motion[297]: [4409184:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
motion[297]: [4409184:motion] [NTC] [ALL] motion_startup: Motion 4.0.1 Started
motion[297]: [4409184:motion] [NTC] [ALL] motion_startup: Logging to syslog
motion[297]: [4409184:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
motion[297]: [4409184:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
motion[297]: [1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
meyectl[283]:     INFO: cleanup started
meyectl[283]:     INFO: wsswitch started
meyectl[283]:     INFO: tasks started
meyectl[283]:     INFO: mjpg client garbage collector started
meyectl[283]:     INFO: server started
meyectl[283]:    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
meyectl[283]: /bin/sh: lsb_release: command not found
meyectl[283]:    ERROR:
meyectl[283]: Traceback (most recent call last):
meyectl[283]:   File "/usr/lib/python2.7/site-packages/tornado/web.py", line 1467, in _stack_context_handle_exception
meyectl[283]:     raise_exc_info((type, value, traceback))
meyectl[283]:   File "/usr/lib/python2.7/site-packages/tornado/stack_context.py", line 316, in wrapped
meyectl[283]:     ret = fn(*args, **kwargs)
meyectl[283]:   File "/usr/lib/python2.7/site-packages/motioneye/mediafiles.py", line 369, in poll_process
meyectl[283]:     read_media_list()
meyectl[283]:   File "/usr/lib/python2.7/site-packages/motioneye/mediafiles.py", line 347, in read_media_list
meyectl[283]:     media_list.append(parent_pipe.recv())
meyectl[283]: EOFError
meyectl[283]:    ERROR: 500 GET /picture/1/list/?_=1507392952633&_username=lebox&_signature=f031a148569821d97a13c7ddef14714d729ddc33 (127.0.0.1) 539.25ms

HW Info:
[root@Rasp01 ~]# lshw -short
H/W path  Device  Class       Description
=========================================
                  system      Raspberry Pi 3 Model B Rev 1.2
/0                bus         Motherboard
/0/0              processor   cpu
/0/1              processor   cpu
/0/2              processor   cpu
/0/3              processor   cpu
/0/4              memory      936MiB System memory
/1        usb1    bus         DWC OTG Controller
/1/1              bus         SMC9514 Hub
/1/1/1            generic     SMSC9512/9514 Fast Ethernet Adapter
/1/1/2            input       USB Keyboard
/1/1/3            multimedia  C922 Pro Stream Webcam
/2        eth0    network     Ethernet interface
/3        wlan0   network     Wireless interface

System:
4.9.52-1-ARCH #1 SMP Sat Sep 30 00:40:14 UTC 2017 armv7l GNU/Linux

Thank you for your help

ccrisan, 2017-10-08:
Have you removed the media folder (the one that's configured for your Root Directory in the settings panel)?

SuperBFG7, 2017-10-08:
I have the exact same issue on a fresh installation: everything works except I can't access image or movie galleries.

Raspbery Pi Zero W
4.9.52-1-ARCH #1 SMP Sat Sep 30 01:00:26 UTC 2017 armv6l GNU/Linux

ccrisan, 2017-10-08:
@SuperBFG7 can you please attach your motioneye log file?

SuperBFG7, 2017-10-08:
Sure, attached. Thanks for the fast reply!

I started motioneye, clicked on picture browser 16:20:53 and on movies browser 16:20:57 and stopped it again to have a clean log.
[motioneye_2017-10-08_1620.log](https://github.com/ccrisan/motioneye/files/1366956/motioneye_2017-10-08_1620.log)

ccrisan, 2017-10-08:
What distro is this? Can you please set log level to DEBUG, try again and send me the new log?

andrin55, 2017-10-08:
I just removed some folders and files inside the camera root not the root itself. I'm currently running Arch Linux for ARM

SuperBFG7, 2017-10-08:
Attached.
I started motioneye, clicked on picture browser 16:34:04 and on movies browser 16:34:11 and stopped it again to have a clean log.

I'm on ArchLinuxARM as well...

[motioneye_2017-10-08_1634.log](https://github.com/ccrisan/motioneye/files/1366960/motioneye_2017-10-08_1634.log)

SuperBFG7, 2017-10-08:
@andrin55 did you update any packages more or less at the same time you delete these files?
I think it might be that an update to some of the Arch python packages broke motioneye.

andrin55, 2017-10-08:
Could be... But I updated it yesterday to the newest version with pip2 --upgrade. The error message stayed the same (the one I posted is from after the upgrade).

ccrisan, 2017-10-08:
What python version do you run? Also what does `pip2 freeze` say?

SuperBFG7, 2017-10-08:
I installed it from the arch user repositories, not with pip2

python2-2.7.14-1
python2-jinja-2.9.6-1
python2-pillow-4.3.0-1
python2-pycurl-7.43.0-3
python2-tornado-4.5.2-1

andrin55, 2017-10-08:
> What python version do you run? Also what does pip2 freeze say?
pip2 freeze:
appdirs==1.4.3
backports-abc==0.5
certifi==2017.7.27.1
Jinja2==2.9.6
MarkupSafe==1.0
motioneye==0.37.1
olefile==0.44
packaging==16.8
Pillow==4.3.0
pycairo==1.13.3
pycurl==7.43.0
pyparsing==2.2.0
RPi.bme280==0.1.2
singledispatch==3.4.0.3
six==1.11.0
smbus==1.1
smbus2==0.1.4
team==1.0
tornado==4.5.2

I'm running python2 2.7.14-1

andrin55, 2017-10-08:
I've set the loglevel to debug but that didn't seem to make any difference:

meyectl[20167]:    DEBUG: 200 GET /static/img/modal-progress.gif (127.0.0.1) 6.85ms
meyectl[20167]:    DEBUG: listing pictures for camera 1
meyectl[20167]:    DEBUG: starting media listing process...
motion[20176]: [1:ml1] [DBG] [VID] v4l2_next: 1) vid_source->pframe 2
motion[20176]: [1:ml1] [DBG] [VID] v4l2_next: 2) vid_source->pframe 2
motion[20176]: [1:ml1] [DBG] [VID] v4l2_next: 3) vid_source->pframe 3 vid_source->buf.index 3
motion[20176]: [1:ml1] [DBG] [VID] v4l2_next: vid_source->buf.bytesused 4147200
motion[20176]: [1:ml1] [DBG] [VID] v4l2_next: the_buffer index 3 Address (6fc2c000)
meyectl[20167]:    ERROR:
meyectl[20167]: Traceback (most recent call last):
meyectl[20167]:   File "/usr/lib/python2.7/site-packages/tornado/web.py", line 1467, in _stack_context_handle_except
meyectl[20167]:     raise_exc_info((type, value, traceback))
meyectl[20167]:   File "/usr/lib/python2.7/site-packages/tornado/stack_context.py", line 316, in wrapped
meyectl[20167]:     ret = fn(*args, **kwargs)
meyectl[20167]:   File "/usr/lib/python2.7/site-packages/motioneye/mediafiles.py", line 369, in poll_process
meyectl[20167]:     read_media_list()
meyectl[20167]:   File "/usr/lib/python2.7/site-packages/motioneye/mediafiles.py", line 347, in read_media_list
meyectl[20167]:     media_list.append(parent_pipe.recv())
meyectl[20167]: EOFError
meyectl[20167]:    ERROR: 500 GET /picture/1/list/?_=1507486821760&_username=lebox&_signature=1f8288302bf915fa708349
motion[20176]: [1:ml1] [DBG] [VID] v4l2_next: 1) vid_source->pframe 3
motion[20176]: [1:ml1] [DBG] [VID] v4l2_next: 2) vid_source->pframe 3
motion[20176]: [1:ml1] [DBG] [VID] v4l2_next: 3) vid_source->pframe 0 vid_source->buf.index 0
motion[20176]: [1:ml1] [DBG] [VID] v4l2_next: vid_source->buf.bytesused 4147200
motion[20176]: [1:ml1] [DBG] [VID] v4l2_next: the_buffer index 0 Address (7080b000)
meyectl[20167]:    DEBUG: creating mjpg client for camera 1
meyectl[20167]:    DEBUG: mjpg client for camera 1 connected on port 8081
motion[20176]: [1:ml1] [DBG] [VID] v4l2_next: 1) vid_source->pframe 0
motion[20176]: [1:ml1] [DBG] [VID] v4l2_next: 2) vid_source->pframe 0

ccrisan, 2017-10-08:
This definitely looks like a problem in recent python version(s). This may be a stupid question but did to you reboot or at least restart motionEye after the last system upgrade?

andrin55, 2017-10-08:
Yes I did, multiple times.

roger-zacharczyk, 2017-10-10:
I can confirm the bug. Noticed when sending e-mail notification with attached image. It fails to find the image. Also, I cannot view the gallery (image nor video gallery).
Same config: Arch for ARM, updated to newest packages (both arch and pip)

andrin55, 2017-10-17:
Is there any progress regarding this issue?

AZRadiohead, 2017-10-22:
I've been experiencing this exact issue.  Everything appears to be working with files being created their appropriate camera1, camera2 directories, but when I go to view either Images or Videos I receive the "An error occurred. Refreshing is recommended." messages.  
I'm running several RPI3's and after a week I switched from running Motioneye to MotionEyeOS with Fast Network Cameras.   I installed Ubuntu 17.04 on dedicated Intel machine yesterday and all was working well.  I rebooted and was greeted with the notice that I could update to Ubuntu 17.10... that's when things went south.  After the 17.10 update I was no longer able to view my images from the Fast Network RPi3's.   I went through the MotionEye install guide and UnInstalled everything and cleaned caches and unused packages and rebooted several times and still can not get the Image/Video view to work. 

I've rebooted the Ubuntu server each time I've uninstalled and reinstalled.  I've done all the apt update, upgrade, dist-upgrade, checked for held packages and everything is showing fine with no problems... Motioneye just doesn't want to show me the images it's created.

Below is snip of what I get when I run 'sudo meyectl startserver'.  It's the exact same issue as the OP has listed. 

-----
 WARNING: 403 POST /_relay_event/?_username=admin&event=picture_save&thread_id=1&_signature=19bb3ee8b85facd53f1d4aea4b3a0a75fea9041c (127.0.0.1) 0.75ms
 WARNING: 403 POST /_relay_event/?_username=admin&event=picture_save&thread_id=1&_signature=19bb3ee8b85facd53f1d4aea4b3a0a75fea9041c (127.0.0.1) 0.69ms
 WARNING: 403 POST /_relay_event/?_username=admin&event=picture_save&thread_id=1&_signature=19bb3ee8b85facd53f1d4aea4b3a0a75fea9041c (127.0.0.1) 0.88ms
 WARNING: 403 POST /_relay_event/?_username=admin&event=picture_save&thread_id=1&_signature=63993da4e1a7b10aae856470dd2e8f67edfcfd76 (127.0.0.1) 0.71ms
 WARNING: 403 POST /_relay_event/?_username=admin&event=stop&thread_id=1&_signature=50a22f432f6c34897d0f5a6129d691051629c20a (127.0.0.1) 0.77ms
 WARNING: 403 POST /_relay_event/?_username=admin&event=movie_end&thread_id=1&_signature=bae1bec87a1ff8f3a60e023492458f2703750cab (127.0.0.1) 0.62ms
   ERROR:
Traceback (most recent call last):
  File "/home/spyglass/.local/lib/python2.7/site-packages/tornado/web.py", line 1467, in _stack_context_handle_exception
    raise_exc_info((type, value, traceback))
  File "/home/spyglass/.local/lib/python2.7/site-packages/tornado/stack_context.py", line 316, in wrapped
    ret = fn(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/motioneye/mediafiles.py", line 369, in poll_process
    read_media_list()
  File "/usr/local/lib/python2.7/dist-packages/motioneye/mediafiles.py", line 347, in read_media_list
    media_list.append(parent_pipe.recv())
EOFError
   ERROR: 500 GET /picture/1/list/?_=1508641511029&_username=admin&_signature=d1a2386bea7e0d06d23289cd7a4ec3234df409cc (192.168.0.47) 504.00ms
^C    INFO: interrupt signal received, shutting down...
    INFO: server stopped
    INFO: tasks stopped
    INFO: motion stopped
    INFO: bye!
-----

ccrisan, 2017-10-29:
Fixed via https://github.com/ccrisan/motioneye/commit/64743308a7508e5744c9764a73489e0ed71afd53.

closed 2017-10-29 by ccrisan

andrin55, 2017-11-01:
Thank you it worked!