motioneye-project/motioneye #686: no motion detected running in docker

magnulu, 2017-10-09:
Hi,

First i tried motioneyeos on a rpi 3 - it worked wonders. Got a lot of false positives with frame change threshold at 5% and 10%, but I expected that. 

Now I am running motioneye in a docker container on my headless server- it works. But, motion detection is not working. Nothing even at 0% threshold. 

This is my motion.log:

> [0:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[0:motion] [WRN] [ALL] thread config option deprecated use camera
[0:motion] [NTC] [ALL] config_camera: Processing camera config file thread-1.conf
[0:motion] [NTC] [ALL] motion_startup: Motion 4.0.1 Started
[0:motion] [NTC] [ALL] motion_startup: Logging to syslog
[0:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://10.0.0.51/user=admin_password=tlJwpbo6_channel=1_stream=1.sdp)
[1:nc1] [ALR] [NET] netcam_handler_loop: Camera handler thread [3] started

Running Motioneye 0.37.1, motion 4.0.1, Linux 4.9.0-3-amd64.

Anyone got a clue?

magnulu, 2017-10-10:
Here is a more detailed logfile from motion: http://paste.debian.net/990093/

Strangely, defining 'log_level' in motion.conf doesn't do anything. Defining 'logfile' does. 

I managed to get this debuglog by changing the 'log_level' setting in motioneye.conf, NOT motion.conf. Also, it is a bit misleading (to me) that the log file from motioneye is called motion.log, while motion per default logs to syslog. The devil is in the detail!

JamBalaya56562, 2026-11-07:
Closing, as this stopped after the debug log, with motionEye 0.37.1 and Motion 4.0.1. As you found, Motion's log level follows `log_level` in `motioneye.conf`, since motionEye starts Motion with a matching `-d`. If motion detection still never triggers on a network camera with a current release, please open a new issue with `motion.log` at debug level.

closed 2026-11-07 by JamBalaya56562