motioneye-project/motioneye #690: E: Unable to locate package libavresample2

stfro, 2017-10-12:
Trying to install motioneye on raspbian stretch lite, however when I try to remove libavcodec-extra-56 libavformat56 libavresample2 and libavutil54, things go wrong:

root@raspberrypi:/home/pi# apt-get remove libavcodec-extra-56 libavformat56 libavresample2 libavutil54
Reading package lists... Done
Building dependency tree
Reading state information... Done
Package 'libavcodec-extra-56' is not installed, so not removed
Package 'libavformat56' is not installed, so not removed
Package 'libavutil54' is not installed, so not removed
E: Unable to locate package libavresample2

I already tried to completely fresh installations, I also tried apt-get update, apt-get upgrade, dist-upgrade, nothing helped.

All this on a RPI zero w.

MHU007, 2018-04-01:
I exactly have the same problem. I have to add that I first uninstall motion. During this last operation I have had a message that some files are still present and the system suggest to erase them. I did it.
After that I tried to install motion again:

Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package motion

stfro, 2018-04-02:
@MHU007 I followed this thread and worked eventually:

https://groups.google.com/forum/#!topic/motioneye/wxdFOn2a28M

JamBalaya56562, 2026-11-07:
Closing as solved, as you got it working by following the forum thread you linked. For anyone installing now, the current steps are in the [README on `main`](https://github.com/motioneye-project/motioneye/blob/main/README.md).

closed 2026-11-07 by JamBalaya56562