motioneye-project/motioneye #691: Output in two different folder places.

karl69942, 2017-10-13:
Hi, I really love this program! I'm just kind of confused though as to why I'm getting two different places where files get created, one is /home/user/Mounted/Pistorage/ and the other is /home/user/Mounted/PiStorage/Camera1/ , I don't quite understand why the /home/user/Mounted/PiStorage is having files created in it & why its not only in the /Camera1/ folder.. 
Am I supposed to disable the motion daemon & just have motioneye use it? I'm sorta confused, sorry if this isn't a good place to ask, but I honestly am not sure where else I should ask this. I went through your wiki but it looks like it doesn't cover this.

chrisf4lc0n, 2017-10-27:
What files? Just the videos or videos and pictures?  You can separately specify the folder for the videos and pictures...

JamBalaya56562, 2026-11-09:
As answered, the file names for movies and pictures can include their own subfolders, so files may end up beside the camera folder when one of them points there. With no follow-up since 2017, closing.

closed 2026-11-09 by JamBalaya56562